RIOT-OS/RIOT #4499: handle of l2src_len in gnrc_ndp_rtr_sol_handle

jfischer-phytec-iot, 2015-12-17:
I'm on implementing of the CDC ECM (Ethernet over USB) and I have a few bugs that lead to overflowing of the pktbuf. Maybe someone can give a little clarity.
One problem occurs if a Router Solicitation from Linux host with the code 0 will be received. Then the l2src_len will be set to zero [1] and it leads to an error in [2], "gnrc_netdev2_eth: destination address had unexpected format"  :smile: .

My current solution for testing is to skip _stale_nc in [1] if l2src_len is zero.
How to proceed with the ncache if the l2src_len  is zero?
Should _send in gnrc_netdev2_eth.c release the pkt buffer in error case?

[1] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c#L422
[2] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/link_layer/netdev2/gnrc_netdev2_eth.c#L185

OlegHahm, 2015-12-17:
I tagged this for the release, because it might be a candidate for the known issues list of the release notes.

authmillenon, 2015-12-18:
I think the correct solution would be to not set the link-layer address if no SLLAO was included into the router solicitation (at least this is what I gather what's happening from your description). Resulting from that the entry wouldn't be staled either (because the neighbor cache is only supposed to be staled if the L2 address in the SLLAO *changes*, not if a SLLAO is not present).

OlegHahm, 2016-02-03:
What's the latest here?

authmillenon, 2016-04-20:
@jfischer-phytec-iot ping?

kYc0o, 2016-07-26:
As far as I can see this issue is solved by #4520. Can anyone confirm it?

miri64, 2016-07-26:
I'm not sure.... @jfischer-phytec-iot can you please confirm?

miri64, 2016-10-18:
@jfischer-phytec-iot ping?

OlegHahm, 2017-01-15:
What shall we do here?

miri64, 2017-11-06:
Fixed by #7925

miri64, 2017-11-27:
Both files referred in this issue were replaced by a complete rework in #7925, so I deem this issue resolved.

closed 2017-11-27 by miri64