RIOT-OS/RIOT #4511: pkg: RELIC uses build system of the original source

authmillenon, 2015-12-18:
This makes it impossible to build for multiple platforms without `clean` inbetween.

OlegHahm, 2015-12-18:
Isn't this true for all packages? And building without `-B clean` is IMO not recommendable for any RIOT module.

authmillenon, 2015-12-18:
> Isn't this true for all packages? 

No at least libcoap, tlsf and microcoap I'm very sure use RIOT's build system.

> And building without `-B clean` is IMO not recommendable for any RIOT module.

I thought we fixed that at least when we are in the same application years ago (and up until now I had no problem with that, either).

authmillenon, 2015-12-18:
At least making the CMake target board dependent would already help a lot ;-)

OlegHahm, 2015-12-18:
> No at least libcoap, tlsf and microcoap I'm very sure use RIOT's build system.

Okay, then we have this problem for every package that comes with its own build system, e.g. RELIC or CCN-Lite.

> I thought we fixed that at least when we are in the same application years ago (and up until now I had no problem with that, either).

I think this is (unfortunately) far from being fixed. I constantly run into this problem when I try to build without `-B clean` and change anything in the code base.

authmillenon, 2015-12-18:
> I think this is (unfortunately) far from being fixed. I constantly run into this problem when I try to build without -B clean and change anything in the code base.

I only have problems when I change macros (or the build system changes them due to included modules etc). But then a simple `clean` is sufficient in most cases (as it was also with this problem).

authmillenon, 2015-12-18:
Another problem is, that it has a loooot of output, so much that I can't even scroll back to the issuing of the command (with gnome-terminal default config).

authmillenon, 2015-12-18:
Building with `make buildtests` also puts out this line:
```
/home/martine/Repositories/RIOT-OS/RIOT/Makefile.include:260: target `/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/relic.a' given more than once in the same rule
```

OlegHahm, 2015-12-18:
Increasing stack size to 16kB "solves" the issue.

OlegHahm, 2016-02-03:
For the record: I'm not planning to work on this issue. If someone else volunteers, please rise your hand, otherwise I tend to close as wontfix.

authmillenon, 2016-02-03:
I can do this as soon as I'm done with my master thesis.

haukepetersen, 2016-04-20:
@authmillenon: could you quickly state on the state (having the merged fixes in mind) and tell me if this issue will be fixed for the relase?! Thanks

authmillenon, 2016-04-20:
This issue isn't really fixed as the title is worded, but building is possible due to #5138 now. Also the spamming of the build output is fixed in #5359, so I guess we postpone, but only mark it as a cleanup now.

kYc0o, 2016-07-26:
> I can do this as soon as I'm done with my master thesis.

Maybe for the next release? ;)

kYc0o, 2016-07-26:
Another package affected by this issue is in #5552 which also provides their own building system, which is incompatible with the RIOT one. Maybe we need to standardise the systematic use of the RIOT build system?

jia200x, 2016-07-28:
With OpenThread I had the same problem. If you are compiling the source directly it makes sense to have a lot of flags and options. But for RIOT, it's nice to use the same options as the main build.

I agree with the idea of @kYc0o

kaspar030, 2016-07-28:
> Maybe we need to standardise the systematic use of the RIOT build system?

Could you elaborate on what you mean by that? I understand you propose to require a RIOTized Makefile for these packages?

kYc0o, 2016-07-28:
Yes that's what I meant, to require Makefiles which use the RIOT build system and not the one embedded on remote Makefiles.

OlegHahm, 2016-07-28:
But that would creating a package much more painful.

kYc0o, 2016-07-28:
Yes I think so too... Maybe post in the wiki to be sure that the build system on the remote package is compatible with RIOT's. I try to make an agreement on this because to discover that the original problem was that one in #5552 took really too long, since the compilation an linking was successful but the used instruction set was different.

kaspar030, 2016-07-28:
> Yes that's what I meant, to require Makefiles which use the RIOT build
> system and not the one embedded on remote Makefiles.

I don't think that is necessary. What is necessary, though, is that the
package build system uses all the CFLAGS, LINKFLAGS etc that RIOT passes
to it.

jia200x, 2016-07-28:
I think in some cases it's possible to just use the pkg Makefile, but at least I have had problems like:

- Duplicate variables in Makefiles that made the remote build crash.
- Complex make system with lots of configurations (e.g I was selecting the right target and had non Thumb code in the end). 

Although the second one could be solved finding the right set of instructions, in the end was easier for me to add the RIOT make system to this pkg though patches.

Cheers

miri64, 2016-10-17:
At this point I don't really care about it anymore (maybe silence `cmake`). If somebody cares, please reopen ;-)

closed 2016-10-17 by miri64

miri64, 2016-10-18:
>  (maybe silence cmake)

(this is wish is also expressed in #5359, so closing this issue stays valid)