RIOT-OS/RIOT #4512: pkg: tests: RELIC unittests fail on iotlab-m3

authmillenon, 2015-12-18:
Both when building in one bulk and singular as tests-relic.

authmillenon, 2015-12-18:
(it crashes for samr21-xpro too)

OlegHahm, 2015-12-18:
On SAMR21-xpro:
```
2015-12-18 22:10:14,118 - INFO # main(): This is RIOT! (Version: 2016.03-devel-51-g16e11-fireflower)
2015-12-18 22:10:14,161 - INFO # ...............................................................................................................................................................................................................................
2015-12-18 22:10:14,171 - INFO # crypto_tripledes_tests.test_crypto_tripledes_encrypt (tests/unittests/tests-crypto/tests-crypto-3des.c 49) exp 1 was -1
2015-12-18 22:10:14,172 - INFO # .
2015-12-18 22:10:14,182 - INFO # crypto_tripledes_tests.test_crypto_tripledes_decrypt (tests/unittests/tests-crypto/tests-crypto-3des.c 71) exp 1 was -1
2015-12-18 22:10:14,182 - INFO # .
2015-12-18 22:10:14,193 - INFO # crypto_twofish_tests.test_crypto_twofish_encrypt (tests/unittests/tests-crypto/tests-crypto-twofish.c 53) exp 1 was -1
2015-12-18 22:10:14,193 - INFO # .
2015-12-18 22:10:14,203 - INFO # crypto_twofish_tests.test_crypto_twofish_decrypt (tests/unittests/tests-crypto/tests-crypto-twofish.c 75) exp 1 was -1
```

Looking into it with the debugger attached, it turns out that malloc is failing.

authmillenon, 2015-12-18:
Then what do I wrong, that it crashes for me?

OlegHahm, 2015-12-18:
That malloc fails is not surprising for the SAMR21.

OlegHahm, 2015-12-18:
It crashes here, too. After that.

OlegHahm, 2015-12-19:
I thought I already commented on that somewhere yesterday, but couldn't find it any more. Setting the stack size to something really big (I tried successfully with 16kB), the test is no more crashing on Cortex-M3. Maybe @tfar has an idea how to improve memory usage to something reasonable.

authmillenon, 2015-12-19:
You told me this on Jabber ;)

OlegHahm, 2016-02-03:
No, I commented here: https://github.com/RIOT-OS/RIOT/issues/4511#issuecomment-165908570

authmillenon, 2016-04-20:
I don't see the relic tests anymore. Is this still an issue?

OlegHahm, 2016-04-20:
Became a unittest.

authmillenon, 2016-04-20:
But they are not included when building for e.g. samr21-xpro:

```
BOARD=samr21-xpro make -C tests/unittests/ flash term
make: Entering directory '/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests'
Building application "unittests" for "samr21-xpro" with MCU "samd21".

touch /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/main.c
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/boards/samr21-xpro
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/core
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/samd21
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/cortexm_common
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/sam21_common
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/sam21_common/periph
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/samd21/periph
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/drivers/periph_common
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/drivers/saul
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/base64
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/bitfield
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/bloom
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/cbor
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/checksum
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/color
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/crypto
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/crypto/modes
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/ecc/hamming256
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/embunit
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/fmt
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/hashes
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/inet_csum
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/netopt
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/netapi
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/netif
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/netif/hdr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/netreg
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/icmpv6
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/hdr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/nc
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/netif
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ndp
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ndp/internal
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ndp/node
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/sixlowpan
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/sixlowpan/ctx
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/sixlowpan/nd
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/sixlowpan/netif
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/pkt
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/pktbuf_static
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/routing/rpl/srh
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/gnrc/transport_layer/udp
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/link_layer/ieee802154
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/fib
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ipv4/addr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ipv6/addr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ipv6/ext
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ipv6/ext/rh
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ipv6/hdr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/sixlowpan
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/transport_layer/udp
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/newlib
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/od
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/phydat
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/pipe
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/random
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/random/tinymt32
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/saul_reg
[... 21 lines cut ...]
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-gnrc_udp
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-hashes
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ieee802154
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-inet_csum
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv4_addr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv6_addr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv6_hdr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv6_nc
[... cut ...]
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-sixlowpan
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-sixlowpan_ctx
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-timex
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ubjson
   text	   data	    bss	    dec	    hex	filename
 236416	   1592	  22480	 260488	  3f988	/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/bin/samr21-xpro/unittests.elf
```

authmillenon, 2016-04-20:
Ah, because of #4514 m)

authmillenon, 2016-04-20:
(still an issue)

kYc0o, 2016-07-11:
still an issue?

kYc0o, 2016-07-26:
Ok well, let's move it for the next release.

kYc0o, 2017-10-20:
Well for me it just complains at build time:

```console
/Users/facosta/git/RIOT-OS/RIOT/sys/cbor/cbor.c: In function 'cbor_deserialize_date_time':
/Users/facosta/git/RIOT-OS/RIOT/sys/cbor/cbor.c:769:9: error: implicit declaration of function 'strptime' [-Werror=implicit-function-declaration]
     if (strptime(buffer, format, val) == 0) {
         ^~~~~~~~
cc1: all warnings being treated as errors
```

kYc0o, 2017-10-24:
With #7783 and #7785 it now compiles for me, however it keeps rebooting when I force this test on any ARM platform.

I suggest to change the name of this issue and also state explicitly that the unittest for relic in ARM are disabled by default.

smlng, 2018-01-30:
`tests-relic` is blacklisted for ARM cortex M platforms see https://github.com/RIOT-OS/RIOT/blob/master/tests/unittests/Makefile#L167

smlng, 2018-01-30:
so close this?

miri64, 2018-01-30:
I think blacklisting a test doesn't fix it.

kYc0o, 2018-07-17:
But who's willing to fix it? I'd say we close as memo and we have it in the release notes as a known issue.

miri64, 2018-07-17:
If it is a known issue for the release then this issue shouldn't be closed..

kYc0o, 2018-07-17:
Ok, I just wanted to find a way to pile more efficiently very old issues, but yeah, I agree closing is not a good way to do it. I thought about a special milestone (like `FIXME First` which existed some time ago) which might help.

miri64, 2018-07-17:
Maybe a project instead so they can get some kind of staging?

tcschmidt, 2018-07-17:
@tfar any proposal for resolving the issue?

tfar, 2018-07-18:
In general RELIC works on ARM, if you have enough RAM and flash memory. So maybe it makes more sense to whitelist boards with enough memory or something like that.

aabadie, 2019-09-29:
This one was probably fixed by #10189. I just tested `tests/pkg_relic` on iotlab-m3 and samr21-xpro and both are working.

miri64, 2019-09-29:
Yepp, we should still remove all the hacks that were added to the unittests because of that.

aabadie, 2019-10-16:
Can we consider #12456 and/or #12455 fixed the hacks added to the unittests and close this issue ?

miri64, 2019-10-16:
The linked PRs don't seem related.... So I'm not sure what you are asking.

miri64, 2019-10-16:
However:

> Yepp, we should still remove all the hacks that were added to the unittests because of that.

This happened in 1e0ca7ca7f7fcb56a208645736a95acc88585b52 (#12040), so we can close this issue indeed.

miri64, 2019-10-16:
(It was even part of the release, so I add the milestone, so it can be marked in the release notes as resolved. Are you ok with that @kb2ma)

closed 2019-10-16 by miri64

aabadie, 2019-10-16:
Thanks @miri64. Normally the dump-issues sub command of the release manager script will handle this automatically: it doesn't use the milestone but the last update date to track the closed bugs (known issues).

kb2ma, 2019-10-16:
Sure @miri64, since #12040 is part of 2019.10, then it's fine to mark this issue as resolved with that same release.