RIOT-OS/RIOT #4513: fib_sr: unittests for fib_sr fail sporadically

authmillenon, 2015-12-18:
```
$ CFLAGS="-gdwarf-3 -g" make -C tests/unittests/ tests-fib_sr test
make: Entering directory `/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests'
touch /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/main.c
Building application "unittests" for "native" with MCU "native".

"make" -C /home/martine/Repositories/RIOT-OS/RIOT/boards/native
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/boards/native/drivers
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/core
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/native
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/periph
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/embunit
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/fib
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/universal_address
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/xtimer
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-fib_sr
   text	   data	    bss	    dec	    hex	filename
  65437	    416	  52196	 118049	  1cd21	/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/unittests.elf
RIOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2016.03-devel-25-gab9f0f-beutlin-ipv6_hdr/fix/2015.12/ipv6_addr_dep)
......
fib_sr_tests.test_fib_sr_06_create_sr_with_hops (tests/unittests/tests-fib_sr/tests-fib_sr.c 264) exp 0 was -2

fib_sr_tests.test_fib_sr_06_create_sr_with_hops (tests/unittests/tests-fib_sr/tests-fib_sr.c 290) exp 0 was -2
......
run 12 failures 2
XXX: lpm_set(): LPM_POWERDOWN not implemented
```

Note, that sometimes they'll pass regardless.

OlegHahm, 2015-12-18:
Tried about 35 times on SAMR21-xpro. Could not reproduce.

authmillenon, 2015-12-18:
On what branch did you try (mine was on release branch (or as in the example above a branch forked from the release branch))

authmillenon, 2015-12-18:
Oh and it was on native. Never tested it for any other platform actually, yet.

OlegHahm, 2015-12-18:
IMO the check here: https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/fib/fib.c#L911 doesn't make much sense and should be removed. @BytesGalore, @cgundogan?

This function (fib_sr_entry_add()) adds a new entry, right? Why should we check its lifetime here? And who should initialize its lifetime?

OlegHahm, 2015-12-18:
@BytesGalore, please check if #4519 seems valid to you.

BytesGalore, 2015-12-19:
@authmillenon @OlegHahm wow never bumped this problem I will have a look.
@OlegHahm `fib_sr_entry_add()` adds an entry behind a given one. 
The function can be used at any time by the routing protocol, so it can also be called when the source route is expired. 
The function enables to update existing source routes without the need to be recreate everything for small changes.
Usually, as a routing protocol, you would use `fib_sr_entry_append()` for putting a new source route to the FIB :).

authmillenon, 2015-12-19:
Well they are there and #4519 does not fix this.

authmillenon, 2015-12-19:
But it seems to be related to the `fib_sr_check_lifetime()` function (which in some cases returns `-ENOENT`).

authmillenon, 2015-12-19:
Its due to the `sr_lifetime` member of the given `sr_fib` being somewhere around 1000000. Does this ring a bell @BytesGalore? I can't really set a watchpoint on this, since it seems to be timing related (setting the watchpoint causes the tests always to succeed).

OlegHahm, 2015-12-21:
#4519 would fix it, but is apparently the wrong solution. ;)

cgundogan, 2016-01-04:
got automatically closed by merging #4524. However, as I understand it, this issue still persists.

kaspar030, 2016-01-06:
Fixed by #4585.

authmillenon, 2016-01-07:
Needs to be merged first.

cgundogan, 2016-01-07:
Can be closed as #4585 was merged into master

PR #4524 (2016-01-04): unittests: some fixes to fib_sr tests
This does not fix #4513, but it makes debugging it a lot easier by correcting the usage of embunit: an assert inside a helper function will cause the helper function to return but not finish the test, leading to very confusing outputs. This minimizes the output of the errors given in #4513 to only one error and I was able to give the analysis I gave there. Additionally, this fixes a lot of the broken formatting in the tests.
PR #4519 (2016-01-27): FIB: remove superfluous things
Checking the lifetime of a newly created source route seems pointless to me.

This PR furthermore removes the `now` parameter for the check lifetime function since this function get call `xtimer_now()` itself.

~~Fixes #4513.~~
closed 2016-01-07 by cgundogan