RIOT-OS/RIOT #4518: fib: unnecessary microsecond precision

OlegHahm, 2015-12-18:
The FIB uses microsecond precision to store the lifetime of a route. IMO this milisecond or even second precision should be sufficient and could save us some memory.

BytesGalore, 2015-12-19:
We could reduce the resolution and precision of a source route lifetime, but then we have the risk of diverging views on a topology.
We could have a situation where the routing protocol assumes that source routes still exists and should be updated soon, while the FIB has already pruned them.
I think this is a matter of deciding what is the lesser evil, overhead of precise timestamps or overhead of re-adding source routes and resynchronization of the topology views in the FIB and the routing protocol.

I formerly decided taking the first one since I can make assumptions for the FIB but not the routing protocols which accesses it. For me it was the most comprehensive approach to not rely on routing protocol behaviour.

OlegHahm, 2015-12-21:
I don't understand how the precision of storing the lifetime affects its up-to-dateness. Of course routing protocol (on all nodes) and the FIB should use the same precision.

kaspar030, 2015-12-21:
> We could have a situation where the routing protocol assumes that source routes still exists and should be updated soon, while the FIB has already pruned them.

That would be a problem from the design perspective. Either FIB or the routing protocol should expire routes, not both.

cgundogan, 2015-12-21:
>That would be a problem from the design perspective. Either FIB or the routing protocol should expire routes, not both.

In general, a routing protocol manages it's view on the nework separately from the FIB. The FIB itself can contain
routes from different routing protocols and using it's own expiration mechanism makes sense to me. The routing protocol should however not really depend on the lifetime of a route that is stored in the FIB. So, IMO, the information flow should be uni-directional from the routing protocol => FIB. I guess that's however not the case currently. Our RPL implementation e.g. uses the fibtable als for it's internal representation to save memory, because we would need to double the information. We may need to transition later to separate the routing information base from the forwarding information base, otherwise it won't be possible to use more than 1 routing protocol simultaneously.

BytesGalore, 2015-12-21:
I introduced the lifetime mechanism to allow "fire and forget" FIB entries and to reduce calling the FIB for removing outdated entries. This can be seen as a most basic and simple replacement policy.
A RP can also always choose to not let the FIB handle the lifetime.

@OlegHahm if we set the lifetimes of RP and FIB say to 1 sec we also get a possible skew of 1-epsilon where epsilon is the highest frequency we can update the lifetime of either RP or FIB.
What could happen is that the FIB deletes the entry, gets instantly a query for the deleted destination, fails to find one, triggers the RP to find a route and returns with `-ENOENT`.
I know this is not earthshaking but it lowers the performance of the FIB and disturbs the RP and the topology.

@kaspar030 the FIB only expires the next hop to a given destination, not the complete route.
A RP might have complex policies for maintaining routes and to exchange hops.
The FIB doesn't know and shouldn't know anything more then next-hops to given destinations.
The lifetime is a only "tool" to allow a simple automatic replacement-"strategy".

Usually the RP should take care to update the lifetime in time, since as @cgundogan said we have a uni-directional communication RP=>FIB.
But I see that changing the granularity would only require the RP to update the lifetime > 1-epsilon before it expires to keep in sync, or just ignore/live with the drift.

(sorry for the wall of text)

cgundogan, 2015-12-21:
@BytesGalore In general I am with @OlegHahm regarding the lifetime unit of the FIB. Having a second precision for routes is sufficient IMO, and we also could try to do this configurable on compile time, maybe?

kaspar030, 2015-12-21:
> we also could try to do this configurable on compile time, maybe?

do we know any routing protocol where <1s precision is needed?

cgundogan, 2015-12-21:
RPL uses seconds precision. I quickly looked into the the drafts/RFCs of AODVv2 and OLSRv2 and couldn't find lifetime precisions <1s. So I guess we should be fine actually with a seconds precision

cgundogan, 2016-03-23:
@BytesGalore what is your current opinion on this topic? Any change of heart maybe since the last discussion? (:

BytesGalore, 2016-03-24:
@cgundogan my heart still bleeds on a `s` based beat

OlegHahm, 2016-03-24:
Maybe @tcschmidt, @waehlisch or @emmanuelsearch have an opinion, too.

tcschmidt, 2016-03-24:
> Maybe @tcschmidt, @waehlisch or @emmanuelsearch have an opinion, too.

Yes, of course :+1: 

IMO it's retty simple: We need a lifetime mechanism because of softstates - this is common in many routing protocols.

I've never seen any (L3) routing protocol working on microseconds. We see from Peter's measurements that internal stack processing (on the IoTLab HW) requires up to 1 ms per packet. To this end, no transmission has happened. Hence, any resolution below 1 ms is bound to lower layers. Btw.: Kernel resolution of Linux and Windows has long been 1 ms for a long time (I don't now current values).

Hence anything below 1 ms appears useless. Personally, I would see 100 ms as a useful granularity - maybe this is difficult to explain to developers?

BytesGalore, 2016-03-24:
`us` is defiantly too fine. IMO `s` precision is too coarse while `ms` is a good balance for the lifetime of a FIB entry from a developer perspective, cf. https://github.com/RIOT-OS/RIOT/pull/5158#issuecomment-200701880 .

kYc0o, 2016-07-26:
So there is a consensus on this issue?

waehlisch, 2016-07-26:
Yes, ms.

kYc0o, 2016-07-26:
So it was changed on the FIB?

kYc0o, 2016-07-26:
I found #5158 is not merged so this will be postponed too.

cgundogan, 2016-10-19:
postponing. see https://github.com/RIOT-OS/RIOT/pull/5158#issuecomment-254831032

miri64, 2018-04-17:
Well the FIB still exist and isn't supposed to be removed (it could be and should be at some point implemented as a back-end for the NIB) and it still has unnecessary microsecond precision, so no: this issue is still valid.

bergzand, 2018-04-17:
Ok, thanks for the explanation.

kaspar030, 2018-04-17:
Still I see this as nice-to-have optimization. Closing as memo, feel free to re-open as appropriate.

closed 2018-04-17 by kaspar030