RIOT-OS/RIOT #4522: avsextrem: linker sometimes doesn't find `bl_init_clks()`

authmillenon, 2015-12-19:
This is the reason, why sometimes `avsextrem` builds failed on travis (back when we build everything far more often this was more often the case than now).

OlegHahm, 2015-12-21:
I cannot remember that I ever saw that on one of my machines. I just built `default` example for avsextrem succesfully about 20 times in a row and `make buildtest` also succeeded several times. Maybe a bug in some toolchain version?

authmillenon, 2015-12-21:
I used the same one you probably did: The one provided with Arch. And as stated above: this happens on Travis from time to time also. 20 times is way to few. With dist/tools/compile_test/compile_test.py there are about 1-2 builds that fail sometimes on my machine.

authmillenon, 2015-12-21:
It's also weird, that this is not happening with `msba2`.

OlegHahm, 2015-12-21:
How often do you think I need to run `dist/tools/compile_test/compile_test.py` until this error should occur?

RIOT-notifier, 2015-12-21:
For both past test cycles I had it 2 times.

OlegHahm, 2015-12-21:
I ran it three times now without this error. Will continue trying.

OlegHahm, 2015-12-23:
After several more runs of the script I'm still unable to reproduce.

authmillenon, 2015-12-23:
Then we just need to be on the look-out for that, I guess.

authmillenon, 2015-12-23:
btw

```
[martine@gruber (tmp)*RIOT]<3 arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 5.3.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

OlegHahm, 2015-12-23:
The computer I tested with has 5.2.0. I'll try upgrading.

miri64, 2016-10-18:
Anyone encounter this problem in the last time? I hadn't had it for a while.

OlegHahm, 2017-01-15:
I think it seems to be save to close it.

closed 2017-01-15 by OlegHahm