RIOT-OS/RIOT #4527: 6lo: Multicast is not forwarded if routing node listens to the address

DipSwitch, 2015-12-20:
Currently if you send a broadcast to ff05::1 for example a routing node doesn't forward the packed if the router itself is listening on the address.

I'll create a pull-request later today to beter illustrate what I mean. :) However thoughts on the subject are welcome :)

DipSwitch, 2015-12-20:
Closed in favor of #4528

authmillenon, 2015-12-20:
Normally we connect an issue and a pull request by commenting "Fixes #<issue>". This way GitHub closes the issue as soon as it is fixed (i.e. the PR was merged.)

miri64, 2016-10-18:
#4528 was adapted in https://github.com/RIOT-OS/RIOT/pull/5729, but it seems not seem to fix this issue...

bergzand, 2018-04-16:
@miri64 Is this with the current implementation still an issue?

miri64, 2018-04-17:
Did not verify :-/

miri64, 2019-08-10:
Mh... this somewhat contradicts https://github.com/RIOT-OS/RIOT/issues/5230... I will try to figure out which one is true and fix.