RIOT-OS/RIOT #4530: MSBA2 MCI Driver timer issue

x3ro, 2015-12-21:
I've run into an odd problem where a long-running operation (block erase) on the MSBA2s SDcard interface fails. If I'm not mistaken, the problem lies [on this line of code](https://github.com/RIOT-OS/RIOT/blob/master/cpu/lpc2387/mci/lpc2387-mci.c#L480), the `xtimer_now()` invocation below. According to the function description, the `tmr` value should have millisecond resolution. Given that `xtimer` provides microsecond resolution, the factor being applied should be `1000` instead of `100`, right? :confused: 

    [...]
    /**
     * @param tmr   Timeout in unit of 1ms
     * @returns 1 when card is tran state, otherwise returns 0
     */
    static int wait_ready(unsigned short tmr)
    {
        unsigned long rc;

        //                                       vvv wrong factor?
        uint32_t stoppoll = xtimer_now() + tmr * 100;
        bool bBreak = false;
    [...]

BytesGalore, 2016-01-26:
right

A-Paul, 2016-01-26:
Agree. Should be 10^3.

OlegHahm, 2016-01-27:
By using `MS_IN_USEC` this wouldn't have happened...

PR #5785 (2016-08-30): lpc2387: Fix wrong factor in driver poll-timer
Fixes #4530
commit 1cf1f59c (2016-08-30): lpc2378: Fix wrong factor in driver poll-timer

Fixes #4530
closed 2016-08-30 by BytesGalore