RIOT-OS/RIOT #4543: FIB: default route not found in FIB

authmillenon, 2015-12-23:
When testing #4144 with a border router setup, I set up the interfaces as follows:

```
ifconfig
2015-12-23 15:00:24,042 - INFO # Iface  5   HWaddr: 65:b2  Channel: 26  Page: 0  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: 4 
2015-12-23 15:00:24,048 - INFO #            Long HWaddr: 5a:55:68:6c:25:35:65:b2 
2015-12-23 15:00:24,054 - INFO #            AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  RTR_ADV  IPHC  
2015-12-23 15:00:24,057 - INFO #            Source address length: 8
2015-12-23 15:00:24,059 - INFO #            Link type: wireless
2015-12-23 15:00:24,064 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2015-12-23 15:00:24,070 - INFO #            inet6 addr: fe80::5855:686c:2535:65b2/64  scope: local
2015-12-23 15:00:24,076 - INFO #            inet6 addr: ff02::1:ff35:65b2/128  scope: local [multicast]
2015-12-23 15:00:24,081 - INFO #            inet6 addr: ff02::2/128  scope: local [multicast]
2015-12-23 15:00:24,082 - INFO #            
2015-12-23 15:00:24,087 - INFO # Iface  6   HWaddr: 00:04:25:19:1a:3e  State: IDLE
2015-12-23 15:00:24,091 - INFO #            MTU:1500  HL:64  RTR  RTR_ADV  
2015-12-23 15:00:24,094 - INFO #            Source address length: 6
2015-12-23 15:00:24,096 - INFO #            Link type: wired
2015-12-23 15:00:24,102 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2015-12-23 15:00:24,108 - INFO #            inet6 addr: fe80::204:25ff:fe19:1a3e/64  scope: local
2015-12-23 15:00:24,114 - INFO #            inet6 addr: ff02::1:ff19:1a3e/128  scope: local [multicast]
2015-12-23 15:00:24,119 - INFO #            inet6 addr: ff02::2/128  scope: local [multicast]
2015-12-23 15:00:24,120 - INFO #            
```

```
ifconfig 6 add abcd::204:25ff:fe19:1a3e/16
fibroute add :: via fe80::xxxx:xxxx:xxxx:xxxx dev 6 # link local of linux machine
ping6 abcd::xxxx:xxxx:xxxx:xxxx
```

and got ping timeouts. Further debugging showed, that `fib_get_next_hop()` did not find the default route as specified, but (I don't know exactly why) the packets where send over the 6LoWPAN.

OlegHahm, 2015-12-23:
What is `fibroute` saying?

authmillenon, 2015-12-23:
```
fibroute
2015-12-23 14:54:58,239 - INFO # > fibroute
2015-12-23 14:54:58,251 - INFO # Destination                             Flags        Next Hop                                Flags      Expires          Interface
2015-12-23 14:54:58,262 - INFO # ::                                      0x00000000 H fe80::xxxx:xxxx:xxxx:xxxx               0x00000000 NEVER            6
```

cgundogan, 2015-12-23:
I think your flags for the fib entry are wrong. `FIB_FLAG_NET_PREFIX` must be set for the `::` route, I believe.

cgundogan, 2015-12-23:
It should be sufficient to set only the destination flags of the fib entry

authmillenon, 2015-12-23:
how do I do that with the `fibroute` command?

cgundogan, 2015-12-23:
`usage: fibroute add <destination> via <next hop> [dev <device>] [lifetime <lifetime>]`

cgundogan, 2015-12-23:
hmm, doesn't seem to work from within the shell then

authmillenon, 2015-12-23:
Okay, with the flag set (I set it statically in the command), it is at least not send over 6Lo anymore and the FIB seems to find the entry. I still get timeouts, but they are probably related to a broken ND.

authmillenon, 2015-12-23:
When I set the next hop in the fib to the GUA of the linux host it works. Fine for me :-)

authmillenon, 2015-12-23:
(the back-translation of 6Lo-ND of ll addresses is the problem)

cgundogan, 2016-01-06:
@BytesGalore I also encountered this problem several times. If you add the default route via the `fibroute` shell command then it is not possible to ping6 between two nodes. I assume it has to do with the destination flag of the default route entry. Because RPL sets the destination flag of the default route to be `FIB_FLAG_NET_PREFIX`. Pinging works then.

OlegHahm, 2016-01-06:
Is this flag new? I remember that default routes used to work last summer.

BytesGalore, 2016-01-06:
@cgundogan yes the problem is that the `FIB_FLAG_NET_PREFIX` is not set by the `fibroute` command, the capital `H` behind the flags indicates the destination is flagged as a host.

cgundogan, 2016-01-06:
@BytesGalore what would be the best solution to solve this? Is it possible to infer from the given ipv6 address if the `FIB_FLAG_NET_PREFIX` should be set automatically (somehow without making the fib specific to IPv6)

BytesGalore, 2016-01-06:
Try #4584 :)

BytesGalore, 2016-01-06:
At least it will allow to set the default route using the `fibroute` command.
With #4279 there will be the proper way to set the prefix.

cgundogan, 2016-01-06:
fixed by #4584

closed 2016-01-06 by cgundogan