RIOT-OS/RIOT #4548: gnrc: higher layer thread starves if there are many receiving packets constantly

Yonezawa-T2, 2015-12-25:
GNRC threads for lower layer of the network stack have higher priority. This may cause starvation of higher layer if there are a lot of receiving packets constantly.

## Option 1, yielding to lower priority threads

For example, calling `gnrc_netapi_get` for higher layer blocks until the higher layer (low priority) thread consumes all queued messages.

~~~~C
/* queue packet for higher layer */
gnrc_netapi_dispatch_receive(GNRC_NETTYPE_IPV6, GNRC_NETREG_DEMUX_CTX_ALL, pkt);

/* wait for higher layer to consume it */
gnrc_netreg_entry_t *sendto = gnrc_netreg_lookup(GNRC_NETTYPE_IPV6, GNRC_NETREG_DEMUX_CTX_ALL);

while (sendto) {
    gnrc_netapi_get(sendto->pid, NETOPT_STATE, 0, NULL, 0);

    sendto = gnrc_netreg_getnext(sendto);
}
~~~~

This may works occasionary (e.g. between 6lo frag and ipv6) but in genral, results in deadlock (e.g. between xbee and ipv6).

## Option 2, separate threads for receiver and sender

To minimize packets stuck in the network stack, higher layer should process a receiving packet as soon as it becomes available while lower layer should send a packet as soon as possible. This whould be achieved by splitting network threads into receiver threads and sender threads like this:

~~~~~
↑ lower priority
L2 receiver
L3 receiver
L4 receiver
...
L4 sender
L3 sender
L2 sender
↓ higher priority
~~~~~

This may work well but imposes memory consumption for stack and message queue as well as code complexity.

Any other options? Or is such usage out of the scope of RIOT?

OlegHahm, 2015-12-25:
Have you observed this problem in practice? I would assume that in most cases the incoming packet rate on low-power radios is magnitudes slower than the MCU speed and processing packets can be done much faster by upper layers than IRQs are raised by the transceiver. However, I agree that this might be a fundamental design issue. But I think neither the proposed option 1 nor 2 are viable. Maybe adapting option 2 to have separate message queues for packets traveling up and down the stack as proposed by @tcschmidt some time ago might fix this. However, with the current approach and IPC this is not immediately implementable.

tcschmidt, 2015-12-25:
Three comments:
 1) GNRC does not support directional duplexing - this is a severe flaw. 
 2) Starvation or deadlocks or large losses are easily constructed ... so I would not go for arguments like "not relevant in practice". Also, the network stack should work for fast network access, as well. 
 3) Splitting upwards and downwards into separate threads adds overhead and complexity. Layers should better account for liveliness by moving queue entries up and down in (quasi-)parallel. This should only require appropriate code - and double queues, as proposed earlier. The problem remaining then is a flow control on the message queues - as also discussed earlier.

Yonezawa-T2, 2015-12-28:
I found the problem in my experiment to use 6LoWPAN to bridge wired networks. See the first picture in #4489. I have tried to transfer file over TCP/IPv6 as a load testing. Many reconstruced 6LoWPAN fragments are discarded by the starvation. 

The test is done on native platform (OS X), where /dev/tty seems to buffer some amount (~3 kB) of data even if termios is configured to send data as soon as possible, so that it might not be problem in embedded platforms. The baudrate was 115,200 bps. `DEBUG` outputs to console also might affect slowdown of entire system.

closed 2019-09-10 by stale[bot]