RIOT-OS/RIOT #4558: What are the requirements on the compiler for a port?

spth, 2015-12-30:
AFAIK, user programs can, but don't have to use C++. How about RIOT itself?
I was wondering if it is possible to port RIOT to a target system that has a C compiler only, no C++. https://github.com/RIOT-OS/RIOT/wiki/Porting-Guide did not tell me.

Philipp

gebart, 2015-12-30:
Short answer: Yes, everything should work with a C-only compiler, except for the C++ examples found in `examples/riot_and_cpp`, `tests/cpp11_mutex`, `tests/cpp11_thread`, `tests/cpp11_condition_variable`.

Possibly the Arduino API-related examples will require C++ as well, I didn't look at the source.

miri64, 2016-10-18:
I think the question is answered or is should we adapt the porting guide?

spth, 2016-10-18:
I'd suggest to add the information to the porting guide.

Philipp

miri64, 2016-10-31:
Updated porting guide

closed 2016-10-31 by miri64