RIOT-OS/RIOT #4560: make: clang is more pedantic than gcc

thomaseichinger, 2015-12-30:
Using `-pedantic` flag with clang compiler, it considers language extensions as errors. Building `tests/fault_handler` for native fails with 
```shell
RIOT/tests/fault_handler/main.c:48:22: error: extension used [-Werror,-Wlanguage-extension-token]
    puts(PRINT_MACRO(INVALID_INSTRUCTION));
                     ^
RIOT/tests/fault_handler/main.c:30:29: note: expanded from macro 'INVALID_INSTRUCTION'
#define INVALID_INSTRUCTION asm (".short 0xdead, 0xbeef, 0xcafe, 0xbabe\n")
                            ^
RIOT/tests/fault_handler/main.c:49:5: error: extension used [-Werror,-Wlanguage-extension-token]
    INVALID_INSTRUCTION;
    ^
RIOT/tests/fault_handler/main.c:30:29: note: expanded from macro 'INVALID_INSTRUCTION'
#define INVALID_INSTRUCTION asm (".short 0xdead, 0xbeef, 0xcafe, 0xbabe\n")
```

thomaseichinger, 2015-12-30:
Also when building `tests/nhdp`
```shell
RIOT/pkg/oonf_api/oonf_api/src-api/common/container_of.h:50:9: error: keyword is hidden by macro definition [-Werror,-Wkeyword-macro]
#define typeof(x) __typeof__(x)
        ^
```

kYc0o, 2016-07-26:
Since it was partially solved by #4593 I'll postpone this issue for the next release.

aabadie, 2017-11-06:
Is this one still valid ?

smlng, 2018-01-26:
since back a while we have `-Wextra` enabled and decided not to go for "-Wpedantic` then. But, as I work on macOS I can confirm that clang is (was GCC 7 is similar) more pedantic even for `-Wextra` - and those errors are mostly fixed by now, too. So I think this is not a big issue anymore.

kYc0o, 2018-01-29:
@smlng I'd say so. We can close this issue.

closed 2018-01-30 by kYc0o