RIOT-OS/RIOT #4561: native: more periph timers

OlegHahm, 2015-12-30:
Currently native supports only one periph timer using POSIX interval timers. In order to run, e.g., the OpenWSN TiSCH implementation on native, more timers are required.

kaspar030, 2015-12-30:
@OlegHahm So xtimer is not accurate enough? Are you sure about that? The overhead of xtimer on native is waaay less than native's periph timer inaccuracy (caused by linux' scheduling).

OlegHahm, 2015-12-30:
The current implementation needs to reset the timer, that's why it uses a dedicated periph timer.

kaspar030, 2015-12-30:
> The current implementation needs to reset the timer

Could you elaborate? Maybe we can improve the API of xtimer.

OlegHahm, 2015-12-30:
Maybe @thomaseichinger can.

haukepetersen, 2016-01-04:
@OlegHahm: the `timer_reset` function is not part of the periph interface anymore. Seems that is also a problem for OpenWSN?

miri64, 2016-10-18:
Soooooo how is the situation?

OlegHahm, 2017-01-15:
I still think that multiple periph timers would be a nice feature for native, but the urgent need for this feature vanished for now.

closed 2020-03-13 by stale[bot]

kaspar030, 2020-03-13:
IMO there's no need for this, applications should use high-level timers (xtimer). The use case for direct periph timers (very high precision timers) is moot on native anyways, no software timer can beat the host scheduler...