RIOT-OS/RIOT #4565: kinetis/rtc: move to periph_common?

basilfx, 2016-01-04:
It seems that [kinetis/rtc](https://github.com/RIOT-OS/RIOT/blob/master/cpu/kinetis_common/rtc.c) is generic and reusable for parts with only an RTT. I need the same implementation for the efm32 port. Would it be an idea to move it to drivers/periph_common, in a similar way it was done for the generic SPI methods?

I could create a PR if needed.

gebart, 2016-01-04:
Good idea!

kaspar030, 2016-01-04:
+1!

Please see  #2538. Maybe we could even move it into sys/? Or drop it completely in favour of a simple epoch-like "real time" and rely on standard C library functions?

IMHO the whole "struct tm" based interface is way to inefficient for our little machines...

basilfx, 2016-01-04:
I created a PR for now. Since it was not much work I don't care what happens with it :-)

miri64, 2016-10-18:
Since #4572 was closed I guess this is still an issue?

gebart, 2016-10-18:
Yes

closed 2019-09-10 by stale[bot]