RIOT-OS/RIOT #4583: cpp11: clang doesn't allow `mutex_t` to be used with `constexpr`

thomaseichinger, 2016-01-06:
All `cpp11-*` tests fail for OS X with clang with the following errors.

```shell
In file included from RIOT/tests/cpp11_condition_variable/main.cpp:26:
RIOT/sys/cpp11-compat/include/riot/mutex.hpp:47:45: error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
  inline constexpr mutex() noexcept : m_mtx{0, PRIORITY_QUEUE_INIT} {}
                                            ^
                                            {}
RIOT/sys/cpp11-compat/include/riot/mutex.hpp:47:20: error: constexpr constructor never produces a constant expression
      [-Winvalid-constexpr]
  inline constexpr mutex() noexcept : m_mtx{0, PRIORITY_QUEUE_INIT} {}
                   ^
RIOT/sys/cpp11-compat/include/riot/mutex.hpp:47:44: note: non-literal type 'mutex_t' cannot be used in a constant expression
  inline constexpr mutex() noexcept : m_mtx{0, PRIORITY_QUEUE_INIT} {}
```
Compiled with
```shell
 % clang -v
Apple LLVM version 7.0.0 (clang-700.0.72)
Target: x86_64-apple-darwin15.2.0
Thread model: posix
```

BytesGalore, 2016-01-06:
strange since it follows exactly the reference definition [1],
But from what the output says I guess we need to provide a custom deleter for the non-literal type of `mutex_t`

@josephnoir could you have a look too

[1] http://en.cppreference.com/w/cpp/thread/mutex/mutex

josephnoir, 2016-01-06:
This makes the compiler happy, but breaks the standard:

```cpp
inline mutex() noexcept : m_mtx{ATOMIC_INIT(0), PRIORITY_QUEUE_INIT} {}
```

Would be nice to find a solution that works with `constexpr`.

josephnoir, 2016-01-06:
Seems a constexpr needs [literal types](http://en.cppreference.com/w/cpp/concept/LiteralType) which must be non-volatile. Changing the integer in `mutex_t` to an `atomic_int_t` broke the `constexpr` constructor. Or am I missing something?

BytesGalore, 2016-01-06:
But
``` C
m_mtx{0, PRIORITY_QUEUE_INIT} {}
```
is a scalar type of literals which is considered a literal [2], and in turn fulfils the requirement for [1].

[1] http://en.cppreference.com/w/cpp/language/constexpr
[2] http://en.cppreference.com/w/cpp/concept/LiteralType

josephnoir, 2016-01-06:
I am with you. After some more reading and consulting colleagues, I too believe this should work. I'll try to create a minimal example and ask the clang guys.

What does this mean for the code -- leave it as is because it is not our fault? Removing the `constexpr` sounds like a bad idea.

BytesGalore, 2016-01-06:
+1 to leave the code as is.
Then we should calm the compiler removing the `-Winvalid-constexpr` check to get rid of the 2 latter warnings.

josephnoir, 2016-01-06:
Filed a bug in the [clang bugzilla](https://llvm.org/bugs/show_bug.cgi?id=26043).

josephnoir, 2016-01-07:
According to the comments [volatile members are no POD data types](http://www.open-std.org/jtc1/sc22/wg21/docs/cwg_defects.html#496) and hence do not count as literal types.

Interestingly enough clang accepts the code if we use the flags `-std=c++14` in `cpp11-compat/Makefile` and the `Makefile` of the testcase.

BytesGalore, 2016-01-07:
@josephnoir I believe c++14 allows static initializations the way we do it

thomaseichinger, 2016-01-07:
@BytesGalore @josephnoir 
So how do we proceed with this? As far as I get the bottom line, this code will be not compliant to standard until c++17 and only mistakenly accepted in c++14 mode. For me this indicates we should change it to be compliant to c++11. (Especially as the module is called cpp11-compat)

josephnoir, 2016-01-07:
The problem is that removing the `constexpr` from the constructor of the mutex is not a good idea. The mutex should be initialized prior to other constructors as those may try to use the mutex.

thomaseichinger, 2016-01-07:
Is there a way to achieve this without `constexpr`?

josephnoir, 2016-01-07:
Don't know.

Other question: why do we need the volatile on the `atomic_int_t? Found [this article](https://www.kernel.org/doc/Documentation/volatile-considered-harmful.txt) on the use of `volatile` in the Linux kernel.

gebart, 2016-01-07:
After a brief scan of the linked article I can agree that volatile may not be necessary here. I have a habit of adding volatile to any variables accessed from both ISR and thread code after some hacky student projects at the university, but I understand that it might not be necessary.

Does anyone here have some more knowledge about volatile?

josephnoir, 2016-01-07:
Sadly not, otherwise I wouldn't have stumbled upon the article looking for information.

BytesGalore, 2016-01-07:
Me neither.
I don't think we can achieve an easy and suitable solution for it quickly.
I guess changing the `atomic_int_t` to non-volatile requires a lot of testing to be certain if nothing breaks.
I'm afraid that we need to postpone fixing this issue from this release :(

thomaseichinger, 2016-01-07:
Ok, so we take this as known issue into the release notes.
Question is, who feels comfortable enough with c++ to come up with a solution?

cgundogan, 2016-01-07:
I mentioned this bug in the `kown issues` section of the release notes

BytesGalore, 2016-01-08:
> Question is, who feels comfortable enough with c++ to come up with a solution?

I don't feel like being the best candidate, but I can try

josephnoir, 2016-01-08:
I am definitely there for the discussion and still think we should evaluate if `volatile` is necessary for the atomic int implementation.

BytesGalore, 2016-01-08:
@josephnoir nice :) teaming up will accelerate finding a proper solution.
> we should evaluate if `volatile` is necessary for the atomic int implementation.

+1, I think this should be the first attempt to check if the issue can be solved without braking other things.

gebart, 2016-01-08:
@josephnoir 
>I am definitely there for the discussion and still think we should evaluate if `volatile` is necessary for the atomic int implementation.

I agree, needs more knowledge though.

Some links:
https://gcc.gnu.org/onlinedocs/gcc/Volatiles.html
http://stackoverflow.com/questions/6397662/if-volatile-is-useless-for-threading-why-do-atomic-operations-require-pointers
http://stackoverflow.com/questions/2484980/why-is-volatile-not-considered-useful-in-multithreaded-c-or-c-programming/2485177#2485177
http://stackoverflow.com/questions/19965076/gcc-memory-barrier-sync-synchronize-vs-asm-volatile-memory

After reading some of the answers I'm starting to lean towards volatile is useful here **in our current implementation**. We don't want the atomic_int_t to be cached in a register, it must be updated in memory. But I guess that since `atomic_cas` is a function (and not inline) the memory will have to be written before returning.
Also we should evaluate if we need to add memory barriers to the mutex family of functions, since they are used to guarantee exclusive access to shared memory. This would prevent the compiler from ever reordering any memory operations across the barrier. It may also be a nonexisting problem.

See if you can find out anything about how other open source OS'es implement their atomic variables.

josephnoir, 2016-01-08:
Found two more interesting sources, one seems to be specific to embedded devices. First one here is the [readme](https://github.com/torvalds/linux/blob/53528695ff6d8b77011bc818407c13e30914a946/Documentation/atomic_ops.txt) from the Linux kernel on atomic implementations and with a bit more code examples a blog entry about [volatile in embedded devices](http://blog.regehr.org/archives/28), see section 6 on the implementation of atomicity. [Here](https://www.cs.utah.edu/~regehr/papers/plos06b.pdf) is a paper about it, from the same guy(?) on the atomic implementation in nesC / TinyOS.

It's a few years old, but they argue in favor of memory barriers instead of volatile.

BytesGalore, 2016-01-08:
The articles @josephnoir found, especially the examples in [1] suggest to me that volatile is probably not best suiting or even not required for our atomic type.
What I understand from the articles so far is that `volatile` helps whenever the compiler (falsly) detects during deploying its call-graphs that variables are not touched, i.e. no write happens, between calls.
Then the compiler may reorder or optimize out variables. But this for my understanding only applies for register mapped variables, where the compiler assumes on compile time a value never changes.
Please correct me if I'm wrong, but this not possible for our atomic int [2] since its in RAM when used, i.e. on the stack.
The only harmful action I can imagine is that the compiler, again falsely reorders read/write calls, where barriers @gebart mentioned would come in handy.

[1] http://blog.regehr.org/archives/28
[2] https://github.com/RIOT-OS/RIOT/blob/master/core/include/atomic.h#L32

josephnoir, 2016-01-11:
Hi. Not sure if it is this easy, but would barriers work in [this way](https://github.com/josephnoir/RIOT/commit/0851d3a4d90f0cf72f3b99ec8a3c686983449b62)?

BytesGalore, 2016-01-11:
At least they are barriers :)
I think you achieve with the two barriers that `mask` cannot be reordered with the assignments/compares, and this way protect the interrupt context order.
Don't know if this is what you intended.

Found a blog referring about barriers [1]
[1] http://preshing.com/20120515/memory-reordering-caught-in-the-act/

josephnoir, 2016-01-27:
Btw, should we open a separate issue for the atomic thing?

BytesGalore, 2016-01-27:
Si Se­ñor.

kYc0o, 2016-07-26:
Someone has a hint on the status of this issue? I'll postpone it but I don't know if there are proposed solutions.

josephnoir, 2016-07-26:
I hope that #5688 will solve this.

gebart, 2016-07-26:
`mutex_t` doesn't actually use `atomic_int_t` anymore (since #4557).
@josephnoir could you test if this issue was fixed by that change?

josephnoir, 2016-07-26:
Thanks for the hint. Compiles on native for me using macOS 10.12 Beta with:
```
Apple LLVM version 8.0.0 (clang-800.0.33.1)
Target: x86_64-apple-darwin16.0.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```
Unless someone wants to try it with the current OS X version, I'd say we can close this.

gebart, 2016-07-26:
Works fine on Linux with `CC=clang` as well.

```
clang version 3.8.1 (tags/RELEASE_381/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```
Let's close this.

closed 2016-07-26 by gebart