RIOT-OS/RIOT #4597: tests/xtimer_hang: test crashes on startup

BytesGalore, 2016-01-06:
Building and executing the test on `native` let the test start and crashes just after printing 
```
...
If it stops before, the test failed.

RIOT/tests/xtimer_hang/../../Makefile.include:286: recipe for target 'term' failed
make: *** [term] Speicherzugriffsfehler (Speicherauszug erstellt)
```
however for @cgundogan the test works (gcc 5.3.0)
for me not (gcc 4.9.2) same for @authmillenon and @haukepetersen (both gcc 4.8.4)

authmillenon, 2016-01-06:
I'm in a quantum state. On observation I'm either an Arch user and it works (same tool chain as @cgundogan) or I'm a Ubuntu Precise user and it fails ;).

Kijewski, 2016-01-06:
Try swapping the binaries between Arch and Ubuntu to tell whether there is a linktime or runtime error.

kaspar030, 2016-01-06:
Could you try to compile with CFLAGS=-g, then run in gdb? (those with non-working test only)

authmillenon, 2016-01-06:
It's a runtime error. I only get it, when executing the binary. No linking involved ;)

authmillenon, 2016-01-06:
@kaspar030 don't have it available right now, but nothing conclusive, it `xtimer_usleep()` in both threads is called, then the segfault occurs with unknown pc.

authmillenon, 2016-01-06:
@Kijewski: remember: no dynamic linking in RIOT (unless native libc is dynamically linked).

Kijewski, 2016-01-06:
@authmillenon, huh? Of course RIOT's native elfs are dynamically linked:
```sh
$ ldd /home/kijewski/RIOT/RIOT/examples/hello-world/bin/native/hello-world.elf
	linux-gate.so.1 (0xf77d3000)
	libm.so.6 => /lib/i386-linux-gnu/i686/cmov/libm.so.6 (0xf7752000)
	libdl.so.2 => /lib/i386-linux-gnu/i686/cmov/libdl.so.2 (0xf774d000)
	libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xf7595000)
	/lib/ld-linux.so.2 (0xf77d6000)
```

authmillenon, 2016-01-06:
Kk then I try what you proposed, as soon as I have bot my machines at hand (any minute now).

authmillenon, 2016-01-06:
First the infodump for @kaspar030:

```
Reading symbols from /home/martine/Repositories/RIOT-OS/RIOT/tests/xtimer_hang/bin/native/xtimer_hang.elf...done.
(gdb) b timer_func1
Breakpoint 1 at 0x804b731: file /home/martine/Repositories/RIOT-OS/RIOT/tests/xtimer_hang/main.c, line 43.
(gdb) b timer_func2
Breakpoint 2 at 0x804b745: file /home/martine/Repositories/RIOT-OS/RIOT/tests/xtimer_hang/main.c, line 52.
(gdb) run
Starting program: /home/martine/Repositories/RIOT-OS/RIOT/tests/xtimer_hang/bin/native/xtimer_hang.elf 
RIOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2016.03-devel-64-ge368-beutlin-HEAD)
xTimer hang test

This test will print "Testing... (<percentage>)" every 100ms for 10 seconds.
If it stops before, the test failed.

Breakpoint 1, timer_func1 (arg=0x0) at /home/martine/Repositories/RIOT-OS/RIOT/tests/xtimer_hang/main.c:43
43	        xtimer_usleep(1000);
(gdb) n

Breakpoint 2, timer_func2 (arg=0x0) at /home/martine/Repositories/RIOT-OS/RIOT/tests/xtimer_hang/main.c:52
52	        xtimer_usleep(1100);
(gdb) step

Program received signal SIGSEGV, Segmentation fault.
0x057c0383 in ?? ()
(gdb) bt
Python Exception <class 'gdb.MemoryError'> Cannot access memory at address 0x108105: 
#0  0xcd7c0383 in ?? ()
Cannot access memory at address 0x108105
(gdb) 
```

authmillenon, 2016-01-06:
The Ubuntu binary on Arch crashes at the same point, but the pc is different (as it is with every execution btw).

The Arch binary on Ubuntu crashes, too. But there is actually some testing happening though the behaviour is quite racy. I get some percentage points and then again a segfault in some unknown regions:

```
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/martine/xtimer_hang.elf 
RIOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2016.03-devel-69-g2bd19c-gruber-2015.12-branch)
xTimer hang test

This test will print "Testing... (<percentage>)" every 100ms for 10 seconds.
If it stops before, the test failed.

Testing... (1%)
Testing... (2%)
Testing... (3%)
Testing... (4%)
[…]
Testing... (37%)

Program received signal SIGSEGV, Segmentation fault.
0x0000ffff in ?? ()
(gdb) bt
#0  0x0000ffff in ?? ()
#1  0xffffffff in ?? ()
#2  0x41ceffff in ?? ()
#3  0x00108101 in ?? ()
#4  0x9d7c0383 in ?? ()
#5  0x002b7956 in ?? ()
#6  0x0805ffff in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

authmillenon, 2016-01-06:
After some runs I got a somewhat more helpful backtrace:

```
#0  0x7f7fffff in ?? ()
#1  0xffffffff in ?? ()
#2  0x41ceffff in ?? ()
#3  0x00108101 in ?? ()
#4  0xcd7c0383 in ?? ()
#5  0x002be8c2 in ?? ()
#6  0x0805ffff in ?? ()
#7  0x0804c655 in mutex_lock (mutex=0x805c1e4 <stack_timer2+516>) at /home/martine/Repositories/RIOT-OS/RIOT/core/mutex.c:49
#8  0x0804c217 in _xtimer_sleep (offset=1100, long_offset=0) at /home/martine/Repositories/RIOT-OS/RIOT/sys/xtimer/xtimer.c:51
#9  0x0804b464 in xtimer_usleep (microseconds=1100) at /home/martine/Repositories/RIOT-OS/RIOT/sys/include/xtimer.h:487
#10 0x0804b495 in timer_func2 (arg=0x0) at /home/martine/Repositories/RIOT-OS/RIOT/tests/xtimer_hang/main.c:52
#11 0xf7df5c6b in makecontext () at ../sysdeps/unix/sysv/linux/i386/makecontext.S:87
#12 0x00000000 in ?? ()
```

cgundogan, 2016-01-07:
I am just taking a wild guess here: Can anybody try replacing:
```c
#define STACKSIZE_TIMER     (1024)
```
with

```c
#define STACKSIZE_TIMER     (THREAD_STACKSIZE_DEFAULT)
```

Does it still crash with this change?

PR #4606 (2016-01-07): tests:xtimer_hang: use THREAD_STACKSIZE_DEFAULT instead of literal.
Without this, `xtimer_hang` segfaults using clang
```
Apple LLVM version 7.0.0 (clang-700.0.72)
Target: x86_64-apple-darwin15.2.0
Thread model: posix
```

Please test for gcc versions <5.*

Fixes #4597.
closed 2016-01-07 by authmillenon