RIOT-OS/RIOT #4608: tests/xtimer_usleep_until: unstable behaviour

A-Paul, 2016-01-07:
While testing "02 Tests - Task 01" *tests/xtimer_usleep_until* has randon errors.

When started repeatedly, sometimes the test fails with "to large difference". When failing count and difference are also not constant.

Tested on "3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt20-1+deb8u2 (2016-01-02) x86_64 GNU/Linux".

gebart, 2016-01-07:
periph/timer on native has known problems with accuracy as far as I know, likely because of concurrency issues running something in a multitasking system without realtime guarantees.

A-Paul, 2016-01-07:
So it would make sense to blacklist `native` for this test (and maybe related tests also)?

kaspar030, 2016-01-07:
It's possible that #4585 fixes this as it vastly increases the backoff times for native. Would you mind trying?

kaspar030, 2016-01-07:
But I cannot reproduce here (4.2.5-1-ARCH #1 SMP PREEMPT Tue Oct 27 08:13:28 CET 2015 x86_64 GNU/Linux). How often do you have to run the test for it to fail?

A-Paul, 2016-01-07:
> [...] How often do you have to run the test for it to fail?

I had a error rate ~75% on a *AMD Athlon(tm) 64 X2 Dual Core Processor 4600+* (Yes some people still drive such oldtimers :D) and ~50% on a *Intel(R) Core(TM)2 Quad CPU    Q9650  @ 3.00GHz*

> It's possible that #4585 fixes this as it vastly increases the backoff times for native. Would you mind trying?

Hit! With #4585 it's much better. Two fails with 70 runs (<3%), on the Athlon .

So, gebarts assumption is correct and this test sometimes **may** fail on native anyway?

cgundogan, 2016-01-07:
@A-Paul is the conclusion good enough to close this issue?

kaspar030, 2016-01-08:
Well, if this fails on an Athlon dual-core, it probably also fails on Raspberry Pies. I'm not sure if increasing the backoff times is a solution, but currently I can't think of a better one...

It might be possible to switch native to a process-monotonic clock, sacrificing accuracy for stability.

authmillenon, 2016-01-08:
Regardless of this issue (I deem it small enough to be a "known issue", since native is only our testing platform) I open a new RC now.

A-Paul, 2016-01-08:
> @A-Paul is the conclusion good enough to close this issue?

If it is expected, due to real time aspects, that the test can't reach 100% i'm fine with it. I'm just a Reporter not a Judge. :D
It would be helpful if the test somehow considers that expectation. What about the proposal I also made in https://github.com/RIOT-OS/RIOT/issues/4609 using `BOARD_BLACKLIST`?

haukepetersen, 2016-04-14:
@A-Paul, @kaspar030, @authmillenon: does this issue still exist?

A-Paul, 2016-04-14:
@haukepetersen, the test behaviour on native is better now and no one advises blacklisting.
I'm closing it now.

A-Paul, 2016-04-14:
Test timing is more stable and no further activity demanded.

closed 2016-04-14 by A-Paul

haukepetersen, 2016-04-15:
nice, thanks!