RIOT-OS/RIOT #4609: tests/periph_rtc: Test fails on native.

A-Paul, 2016-01-07:
While testing "02 Tests - Task 01" *tests/periph_rtc* following showed up.

> RIOT RTC low-level driver test
This test will display 'Alarm' in 10 seconds
Initializing the RTC driver
Native RTC initialized.
Setting clock to 2011-12-13 14:15:15
/home/aar680/git/RIOT/tests/periph_rtc/bin/native/periph_rtc.elf: rtc_set_time: not implemented
Clock set to 2016-01-07 15:50:13
Setting alarm to 2016-01-07 15:50:23
/home/aar680/git/RIOT/tests/periph_rtc/bin/native/periph_rtc.elf: rtc_set_alarm: not implemented
Alarm set to 2016-01-07 15:50:23
The alarm should trigger every 10 seconds for 4 times.

Then nothing happens...

Seemingly `rtc_set_time` and `rtc_set_alarm` are not implemented for native on purpose. It might be sensible -and less time consuming- to blacklist this test for native.

Would `BOARD_BLACKLIST := native` in *Makefile* be the intended measure?

cgundogan, 2016-01-08:
If the test does not work for native because of missing functionalities, then I would strongly vote for blacklisting it for this test.

kaspar030, 2016-01-08:
Or remove the feature from native? It's incomplete.

OlegHahm, 2016-01-08:
Maybe https://github.com/RIOT-OS/RIOT/pull/4563 can be used to implement RTC alarm.

LudwigKnuepfer, 2016-01-08:
I would rather vote for completing the implementation then removing it, but then I have no time to do this right now.

kYc0o, 2018-04-16:
Right now we have I have this output on OS X:

```console
RIOT native interrupts/signals initialized.
Native RTC initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.07-devel-945-gfbea4-snake.local)

RIOT RTC low-level driver test
This test will display 'Alarm!' every 2 seconds for 4 times
  Setting clock to   2011-12-13 14:15:15
/Users/facosta/git/RIOT-OS/RIOT/tests/periph_rtc/bin/native/tests_periph_rtc.elf: rtc_set_time: not implemented
Clock value is now   2018-04-16 18:01:53
  Setting alarm to   2018-04-16 18:01:55
/Users/facosta/git/RIOT-OS/RIOT/tests/periph_rtc/bin/native/tests_periph_rtc.elf: rtc_set_alarm: not implemented
   Alarm is set to   2018-04-16 18:01:55
```

Should we consider this is enough to close this issue? It's being two years is here and no one implemented the feature, thus I'd say it's not needed, and at the same time it doesn't hurt to have it compiled for native.

closed 2019-09-10 by stale[bot]