RIOT-OS/RIOT #4621: pkg: micro-ecc: test takes 2 hours

OlegHahm, 2016-01-08:
According to https://github.com/RIOT-OS/Release-Specs/issues/14 the `pkg_micro-ecc` test takes about two hours to complete. I remember similar results for the last release. (Though it's much faster (~30 minutes) on my computers.) The main question is why this test takes so long and if this package is suitable at all for IoT hardware if takes such a long time on native. But maybe it is just the test. Anyone with more insights?

d00616, 2016-01-08:
I think you are running out of random numbers while running tests -> cat /proc/sys/kernel/random/entropy_avail

There are two solutions.
 1. Switch to /dev/urandom -> http://www.2uo.de/myths-about-urandom/
 2. Use service like haveged to fill entropy buffer when it's low

In last Month i had problems in various scenarios

d00616, 2016-01-08:
@OlegHahm  Answer of second part of your first question: This and any similar package is insecure on systems without good random numbers. IoT hardware without an RNG source needs an generic solution collecting events, noise... to build unpredictable random numbers. Without this it's impossible generating keys or session keys in the device.

OlegHahm, 2016-01-13:
Thanks for the clarification.

closed 2016-01-26 by cgundogan