RIOT-OS/RIOT #4628: memory overflow in RIOT2015-09 and performance bugs in gnrc_ndp_internal.c

zq301, 2016-01-11:
memcpy API may introduce memory overflow, like the bug in "https://github.com/contiki-os/contiki/issues/707". In the calling process of memcpy, it may cause overflow when the third parameter is a constant, but the the second parameter is a char * type. Because memcpy will not stop when it encounter a ‘\0’, but the length of the second parameter may less than the value of the third parameter. The followings are potential bugs of this type in RIOT-2015.09.
1)	RIOT-2015.09/cpu/native/netdev2_tap/netdev2_tap.c, line 84
memcpy(dev->addr,src,ETHERNET_ADDR_LEN)，
2)	RIOT-2015.09/sys/crypto/chacha.c 79, line
memcpy(ctx->state+4,key,32) 
3)	RIOT-2015.09/sys/crypto/chacha.c, line 83
memcpy(ctx->state+4,key,16)
4)	RIOT-2015.09/sys/crypto/chacha.c, line 84 
memcpy(ctx->state+8,key, 16)

In the file RIOT-2015.09/sys/net/gnrc/network_layer/ndp/internal/gnrc_ndp_internal.c line 623 and line 663, the loop’s function is to Shave off zero from the end of a string. The code is as follows: 

for (; s12a[s12a_len-1]==0x00; s12a_len--);

But if the string is very long, it will decrease the system performance. both functions parse the string byte wise and are quite slow for longer strings may be improved by add the byte wise to 16 bits wise or more In an appropriate way.

kaspar030, 2016-01-11:
> memcpy API may introduce memory overflow, like the bug in "contiki-os/contiki#707". In the calling process of memcpy, it may cause overflow when the third parameter is a constant, but the the second parameter is a char * type. Because memcpy will not stop when it encounter a ‘\0’, but the length of the second parameter may less than the value of the third parameter.

Well, that behaviour is expected and well defined.

> 1) RIOT-2015.09/cpu/native/netdev2_tap/netdev2_tap.c, line 84
> memcpy(dev->addr,src,ETHERNET_ADDR_LEN),

No problem here.

authmillenon, 2016-01-11:
> 1) RIOT-2015.09/cpu/native/netdev2_tap/netdev2_tap.c, line 84
memcpy(dev->addr,src,ETHERNET_ADDR_LEN)

Length of `src` is checked [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/native/netdev2_tap/netdev2_tap.c#L155)

kaspar030, 2016-01-11:
> 2) RIOT-2015.09/sys/crypto/chacha.c 79, line
> memcpy(ctx->state+4,key,32)
> 3) RIOT-2015.09/sys/crypto/chacha.c, line 83
> memcpy(ctx->state+4,key,16)
> 4) RIOT-2015.09/sys/crypto/chacha.c, line 84
> memcpy(ctx->state+8,key, 16)

These seem OK to me, too. @zq301 did you take a loot at the code? The length of the source buffer is explicitly compared against the constants.

authmillenon, 2016-01-11:
> In the file RIOT-2015.09/sys/net/gnrc/network_layer/ndp/internal/gnrc_ndp_internal.c line 623 and line 663, the loop’s function is to Shave off zero from the end of a string. The code is as follows:

> for (; s12a[s12a_len-1]==0x00; s12a_len--);

> But if the string is very long, it will decrease the system performance. both functions parse the string byte wise and are quite slow for longer strings may be improved by add the byte wise to 16 bits wise or more In an appropriate way.

Since this is expected to be an address it is very unlikely, that this "string" is longer that 8 byte.

kaspar030, 2016-01-11:
At least the contiki guys have tons more of these problems ;)

authmillenon, 2016-01-11:
I'm sorry, but none of the issues you pointed out seem to be strings in the `\0` terminated sense, but other array structures of fixed length, that are guarded against overflows by given length parameters. Maybe there is a misunderstanding, but if not I would prefer to close this issue as "invalid" if this is okay for you.

authmillenon, 2016-01-11:
Oops, sorry @kaspar030 the reassignment was by accident… didn't know that even came through :/

authmillenon, 2016-01-11:
But thanks for your comments too :-)

kaspar030, 2016-01-11:
@authmillenon No problemo, I started looking into it before you assigned be. ;)

zq301, 2016-01-12:
@authmillenon @kaspar030 OK, I will read the code more serious.

kaspar030, 2016-01-12:
Then I'll close this for now.

@zq301 Thank you for reporting!

closed 2016-01-12 by kaspar030