RIOT-OS/RIOT #4634: RIOT communication Bug found with Samr21-xplained-pro experiments

zhuosg, 2016-01-13:
Dear all, the attachment is a technical/experimental report for describing the Bug we found with RIOT wireless communication experiment on Samr21-xplained-pro boards.

In short, the bug is a “destination-address” error that will put the wrong destination address into the MAC layer head field when sending out, thus the intended receiver will disregard/drop the packet from the sender, resulting in communication fault.

[RIOT Bug found with Samr21-xplained-pro experiments.pdf](https://github.com/RIOT-OS/RIOT/files/88655/RIOT.Bug.found.with.Samr21-xplained-pro.experiments.pdf)

cgundogan, 2016-01-13:
Could be related to #4215 but I cannot check right now

zhuosg, 2016-01-13:
In the experiment we conducted, the communication stack used the long MAC address, as indicated in the sniffer.

thomaseichinger, 2016-01-21:
@zhuosg What you observed here is mentioned [in this section](https://tools.ietf.org/html/rfc4944#section-6). Namely the change of the "Universal/Local" when constructing the interface identifier.

OlegHahm, 2016-03-09:
@zhuosg, are you okay with @thomaseichinger's explanation?

cgundogan, 2016-03-23:
Did anyone else encounter this problem? The user is not available anymore.

OlegHahm, 2016-03-23:
From the discussion I guess that @zhuoshuguo was the asking user. Am I correct?

zhuoshuguo, 2016-03-23:
Yes, I am the asking user (asked by another account of mine, which has been deleted),  and I am OK  with @thomaseichinger's explanation. Thanks a lot!  Sorry for the late reply (didn't check emails related to the deleted account).

closed 2016-03-23 by OlegHahm