RIOT-OS/RIOT #4644: nrf6310: LED is misconfigured

gebart, 2016-01-15:
Probably something that went wrong in the GPIO API revision. I don't have the board and no energy to put into fixing it for that board. Just reporting so that someone else might take it up later.

```
% make TOOLCHAIN=llvm BOARD=nrf6310
Building application "periph_uart" for "nrf6310" with MCU "nrf51".

"make" -C /home/jgn/work/src/riot/boards/nrf6310
.../src/riot/boards/nrf6310/board.c:29:5: error: array index 256 is past the end of the array (which contains 32 elements)
      [-Werror,-Warray-bounds]
    NRF_GPIO->PIN_CNF[LED_RED_PIN] = GPIO_PIN_CNF_DIR_Output;
    ^                 ~~~~~~~~~~~
.../src/riot/cpu/nrf51/include/nrf51.h:1294:41: note: expanded from macro 'NRF_GPIO'
#define NRF_GPIO                        ((NRF_GPIO_Type           *) NRF_GPIO_BASE)
                                        ^
.../src/riot/cpu/nrf51/include/nrf51.h:1195:3: note: array 'PIN_CNF' declared here
  __IO uint32_t  PIN_CNF[32];                       /*!< Configuration of GPIO pins.                                           */
  ^
.../src/riot/cpu/cortexm_common/include/core_cm0.h:198:21: note: expanded from macro '__IO'
#define     __IO    volatile             /*!< Defines 'read / write' permissions              */
                    ^
1 error generated.

```

gebart, 2016-01-17:
I happened to stumble upon the origin of the problem, it is fixed by https://github.com/gebart/RIOT/commit/4699c9abae0d4e8251b74c23b3c6e5910b54d064

d00616, 2016-01-17:
> gebart@4699c9a

Thanks. This looks fine.

gebart, 2016-03-09:
Fixed by #4828 I assume

closed 2016-03-09 by gebart