RIOT-OS/RIOT #4657: cpu: cc2538: xtimer broken

aeneby, 2016-01-20:
None of the ```tests/xtimer_*``` tests complete on this platform, and using (for example) ```xtimer_usleep``` in an application causes it to hang.

OlegHahm, 2016-01-21:
@thomaseichinger, @haukepetersen, do you have the hardware to test? Is there way to use the OpenMote with a debugger?

haukepetersen, 2016-01-21:
I think there is, though I never tried it... Not sure where our openmote node and the flash are at the moment, @thomaseichinger, do you know?

A-Paul, 2016-01-28:
Just want to confirm. In https://github.com/RIOT-OS/RIOT/pull/4040 the board failed big time!

kaspar030, 2016-01-29:
Could you test the periph/timers without xtimer? (tests/periph_timer)

aeneby, 2016-01-29:
```
2016-01-29 22:31:09,220 - INFO # main(): This is RIOT! (Version: 2016.03-devel-259-g03081-pontus)
2016-01-29 22:31:09,221 - INFO # 
2016-01-29 22:31:09,222 - INFO # Test for peripheral TIMERs
2016-01-29 22:31:09,223 - INFO # 
2016-01-29 22:31:09,227 - INFO # This test will test all configured peripheral timers of the
2016-01-29 22:31:09,233 - INFO # targeted platform. For each timer, it will set each channel with
2016-01-29 22:31:09,239 - INFO # an incrementing timeout. CH0 set to 5ms, CH1 to 10ms, CH2 to 15ms
2016-01-29 22:31:09,240 - INFO # and so on.
2016-01-29 22:31:09,245 - INFO # In the output you should see that every channel fired, after an
2016-01-29 22:31:09,251 - INFO # evenly distributed amount of time -> the shown diff values should
2016-01-29 22:31:09,254 - INFO # be pretty much equal (to some jitter...)
2016-01-29 22:31:09,260 - INFO # This test does however NOT show, if the timeouts were correct in
2016-01-29 22:31:09,266 - INFO # relation to the expected real-time ~ use e.g. tests/xtimer_msg for
2016-01-29 22:31:09,267 - INFO # this.
2016-01-29 22:31:09,267 - INFO # 
2016-01-29 22:31:09,267 - INFO # 
2016-01-29 22:31:09,268 - INFO # Available timers: 4
2016-01-29 22:31:09,269 - INFO # 
2016-01-29 22:31:09,270 - INFO # Testing TIMER_0:
2016-01-29 22:31:09,273 - INFO # TIMER_0: initialization successful
2016-01-29 22:31:09,274 - INFO # TIMER_0: stopped
2016-01-29 22:31:09,277 - INFO # TIMER_0: set channel 0 to 5000
2016-01-29 22:31:09,280 - INFO # TIMER_0: set channel 1 to 10000
2016-01-29 22:31:09,281 - INFO # TIMER_0: starting
```
Then it just kind of hangs out in the do-while loop in test_timer()...

hexluthor, 2016-02-01:
See my PR https://github.com/RIOT-OS/RIOT/pull/4729.

kYc0o, 2016-04-04:
#4729 was merged and the issue is not present anymore, so I'll close it.

closed 2016-04-04 by kYc0o