RIOT-OS/RIOT #4669: core: re-visit `xxx_arch` headers (COREIF_NG)

haukepetersen, 2016-01-22:
While reviewing #3984 I found that re-visiting the concept of the `COREIF_NG` headers (`core/include/arch/xxx_arch.h`) might make sense. The main idea was to clearly mark all dependencies from the core to platform specific code, to make porting more easy and transparent. 

Pros:
- clean separation between core and boards/cpu
- porting (the core) is (theoretically) straight forward: just implement all functions defined in the `core/include/arch/xxx_arch.h` headers and you are done
- one can see easily which functions are platform dependent

Cons:
- some functions are provided by the core to other modules (e.g. `enableIRQ()`), but their implementations are platform dependent and as such would have to follow the `xxx_arch()` naming scheme (e.g. `enableIRQ_arch()`) -> ugly
- the above fact leads to very ugly re-defines of function names, e.g.:
```c
#ifdef COREIF_NG
#define enableIRQ       irq_arch_enable
#define disableIRQ      irq_arch_disable
#define restoreIRQ      irq_arch_restore
#define inISR           irq_arch_in
#endif
```
- just implementing all of the header is not enough for porting, there are on top some defines etc. that also have to be put in the board/cpu code and are not clearly marked anyway
- not all platforms follow this scheme, so at the moment we have at least two different ways we implement the platform specific core interfaces

So I don't have a strong opinion in either way, keep this concept or maybe alter/drop it, but I would strongly urge for consistency!

Alternative: revert the `_arch` approach (by moving the headers back into `core/include` and dropping the `_arch` where appropriate), but create a clean and complete list in the documentation listing all the needed function and defines that a port must supply.

Any other ideas/opinions to this topic?

kaspar030, 2016-01-22:
I'm totally for dropping the _arch approach. The clean seperation comes at the price of unclean code.

> create a clean and complete list in the documentation listing all the needed function and defines that a port must supply.

Yep. Maybe a PORTING.md containing:

- a folder skeleton
- a list of functions that *must* be implemented
- a list of core peripherals that *must* be implemented
- a list of defines that *must* be set
- guidelines regarding vendor libraries
- pointer to reference implementations
- ...

authmillenon, 2016-01-22:
> Yep. Maybe a PORTING.md containing:

how about putting this directly into the doxygen doc instead ;-)

haukepetersen, 2016-01-22:
I don't care where we put it, as long as it is there :-)

haukepetersen, 2018-01-24:
I consider this issue solved with #7781 -> closing it.

closed 2018-01-24 by haukepetersen