RIOT-OS/RIOT #4672: Rethinking network interface auto-initialization

authmillenon, 2016-01-22:
Working on non-GNRC I realized that even most netdev2 devices are auto-initialized very GNRC-dependent so I think it makes sense to make the auto-initialization of network devices more granular.

There are a two things to consider:
* netdev(2) auto-initialization
* netdev(2) => interface mapping for every stack

The first one should be doable pretty straight forward with our current approach (just leave out the GNRC stuff). The second needs to have a default setting for every stack (e.g. `gnrc_nomac` interface for `at86rf2xx` or default `gnrc_netdev2` for `cc11xx`), but should also be easily overridable by an application (if e.g. another mac layer is to be used). One way to do this could be to e.g. could be that every stack needs to define a function. For GNRC this might be:

```C
void gnrc_init_netif(netdev2_t *dev, void *netif_params);
```

where `netif_params` is a stack-related (or even interface-type related) data type that gives the parameters for the actual stack-internal interface initialization function(s). The mapping can then be done to provide a struct type to the auto-initialization.

haukepetersen, 2016-01-22:
not quite sure I understand all of it. On the first part, so initializing just the network device (without any dependencies to any network stack) I agree that our current approach with maybe slight alterations (-> #4658) should do the job.

Leaves the mapping of (pre-initialized) network devices to network stacks and MAC layers, and this seems to be a little more complex to me. I think defining a generic 'network-stack-init-function' prototype, which each network stack has to implement might be the way to go, but not sure how this would look like.

authmillenon, 2016-01-22:
> not quite sure I understand all of it.

Well, that might be, because my ideas are currently scetchy at best ;-). So let's come up with a comprehensive solution together :-).

The current idea was that somewhere(TM) there is a place that a number of structs `ifaces` or something like this would be stored, where the structs would look something like this:

```C
struct {
  void (*init)(netdev2_t *, void *);  /**< one of the init-functions above */
  netdev2_t *dev;                     /**< netdev parameter for init */
  void *iface_params;                 /**< params parameter for init */
};
```

And then be called by the auto initialization in such way:

```py
for i in ifaces:
   # device specific setup and init of i->dev
   i.init(i.dev, i.iface_params);
```

But I currently don't know where something like that could be stored, nor how to provide alternative configurations via the application.

haukepetersen, 2016-01-22:
I see... Looks promising, but I see at least one problem. Let's say we define the contents of this struct in some configuration file as this
```c
const your_proposed_struct_t ifaces[] = {
    {.init = &gnrc_init, .dev = ???, .iface_params = &where_get_these_from},
    {.init = &nanonet_init, .dev = ???, .iface_params = &again_how_to_get_these},
};
```
So now we can not statically map the device descriptors of the network devices to this config array (as they are in RAM and created at run-time. Second problem is to map the right parameters to the right entry here... But if we find a way to solve this we should be on a good path...

authmillenon, 2016-01-22:
There are also macros stuff we can create from parsable files like `yml` to feed the preprocessor...

haukepetersen, 2016-01-22:
True, but if possible I would like to circumvent this (same as @kaspar030 's promising snips). Let's try to find a suitable solution without any buildsystem/linkerscript/weak stuff first... So let's think!

authmillenon, 2016-01-22:
Also I thought it more like this:

```
const your_proposed_struct_t ifaces[] = {
#ifdef MODULE_AT86RFXX
#ifdef MODULE_GNRC
    {.init = &gnrc_init, .dev = &at86rfxx, .iface_params = &gnrc_netif_default},
#endif
#ifdef MODULE_LWIP
    {.init = &lwip_init, .dev = &at86rfxx, .iface_params = &lwip_netif_default},
#endif
#endif
/* ... */
};
```

authmillenon, 2016-01-22:
> True, but if possible I would like to circumvent this (same as @kaspar030 's promising snips). Let's try to find a suitable solution without any buildsystem/linkerscript/weak stuff first... So let's think!

Macros wouldn't be dependent on any of these and 100% C-compatible... But I don't know any solution yet.

authmillenon, 2016-01-22:
Okay, let's break this down, what do we have:
* a network stack,
* a `netdev2`, and
* network stack specific parameters:
  * for GNRC these are the MAC layer thread type and/or the gnrc_netdev2 struct.
  * for lwIP and emb6 this is mainly the input function IIRC

kaspar030, 2016-01-22:
At least at the "device gets it's pin configuration" layer, netdev2 devices should not be different than say, a sensor driver. Network stacks should then collect netdev2 devices a layer higher.
So to go in line with @haukepetersen's consolidating scheme, netdev2 devices should have something like "at86rf_init(handle, params_struct)", same as our other drivers.

kaspar030, 2016-01-22:
> So now we can not statically map the device descriptors of the network devices to this config array (as they are in RAM and created at run-time.

I don't get that. If there's a struct like that, why can't we map from a device descriptor to the struct entry? The struct being in flash doesn't make a difference.

OlegHahm, 2016-03-29:
> So to go in line with @haukepetersen's consolidating scheme, netdev2 devices should have something like "at86rf_init(handle, params_struct)", same as our other drivers.

Without knowing/remembering this issue here, I think #5185 addressed this, right? I'm wondering if we should make the second parameter of the init/setup functions more generic, which might result in something like a void pointer since the parameters are very diverse.

kYc0o, 2016-07-26:
Let's postpone this one since hopefully a newer solution will come based on #5181

PeterKietzmann, 2017-01-26:
Ping @miri64 could you answer @OlegHahm's [comment](https://github.com/RIOT-OS/RIOT/issues/4672#issuecomment-202918197)

miri64, 2017-01-28:
> Without knowing/remembering this issue here, I think #5185 addressed this, right?

Yes.

> I'm wondering if we should make the second parameter of the init/setup functions more generic, which might result in something like a void pointer since the parameters are very diverse.

I was thinking the same, something like a generic `netdev2_params`. But I have to admit that I did not yet think the concept through enough to justify the required grow in RAM to identify the specific type.

closed 2020-03-13 by stale[bot]