RIOT-OS/RIOT #4673: boards: exact same configuration but different board names?

emmanuelsearch, 2016-01-22:
Concrete example: TI LaunchPad recently revamped/renamed its series.

So for instance, the ek-lm4f120xl currently supported by RIOT is now reissued by TI as the ek-tm4c123gxl, with the same specs.

So the question is: how do we deal with that? To avoid code duplication, the same config files should be used wether we make/flash/term for either board names.

haukepetersen, 2016-01-22:
Hm, thinking about it: for the build system something like aliases for board names might solve it. 

But for understanding/finding code in the source tree it does not solve the matter - say I want to look into the code for the `ek-tm4c123gxl` board, but this is matched internally to the `boards/ek-lm4f120xl` folder it might be (very) hard for people to actually locate their board's code?!

Kijewski, 2016-01-22:
```
$ make BOARD=ek-tm4c123gxl

Note: RIOT uses "ek-lm4f120xl" as canonical name for "ek-tm4c123gxl" boards.
Building application "foo" for "ek-lm4f120xl" with MCU "bar".
...
```

OlegHahm, 2016-01-22:
> But for understanding/finding code in the source tree it does not solve the matter - say I want to look into the code for the ek-tm4c123gxl board, but this is matched internally to the boards/ek-lm4f120xl folder it might be (very) hard for people to actually locate their board's code?!

How about symlinks? Apparently this should work even for Windows:
http://stackoverflow.com/questions/11662868/what-happens-when-i-clone-a-repository-with-symlinks-on-windows

kaspar030, 2016-01-22:
> Note: RIOT uses "ek-lm4f120xl" as canonical name for "ek-tm4c123gxl" boards. Building application "foo" for "ek-lm4f120xl" with MCU "bar".

I like.

The aliases themselves could be defined in a flat Makefile.board_aliases, which gets included first thing after the Application's ;akefile.

```
BOARD_ALIAS:=$(BOARD)
ifneq (,$(filter $(BOARD), ek-tm4c123s ek-tm4c123xl ek-tm4c123m))
  BOARD:=ek-tm4c123
endif

(...)

ifneq ($(BOARD), $(BOARD_ALIAS))
$(info Rene's message)
endif
```

kaspar030, 2016-01-22:
> How about symlinks?

Ufffff... all the sloccount, find -name '*.[hc]' -exec, ... magic would break ...

OlegHahm, 2016-01-22:
Horrible!

haukepetersen, 2017-04-25:
Is this still an issue? We have not run into this anywhere else and there is not much traction (currently) for the LaunchPad boards...

bergzand, 2018-04-16:
Would this be fixed with #7976?

emmanuelsearch, 2019-09-03:
This did not turn out to be a problem in the the long run. Closing.

closed 2019-09-03 by emmanuelsearch