RIOT-OS/RIOT #4677: two CoAP examples

Lotterleben, 2016-01-24:
Hi,
RIOT currently has two CoAP examples, one that compiles (https://github.com/RIOT-OS/RIOT/tree/master/examples/microcoap_server) and one that doesn't (https://github.com/RIOT-OS/applications/tree/master/microcoap). This appears to have caused quite the confusion (judging by the comments we got on our watr.li post, which still links to the old example '^^), and, well, one is outdated anyway.
Is there a reason for this? Otherwise, I'd be happy to remove the latter and add some comments to the former's `handle_get_well_known_core()` function and move the tips about firefox copper and aiocoap from the latter's into the former's Makefile.

bapclenet, 2016-01-26:
Yes the example in application is outdated and should be removed. Feel free to move firefox copper and aiocoap from the old to the new implementation.

authmillenon, 2016-01-26:
I'm not sure if it was @OlegHahm or @haukepetersen, but someone stated a (for me) valid reason to keep the application because I wanted to remove it as soon as `microcoap_server` was merged... But sadly, I forgot this reason as well :neutral_face:

Lotterleben, 2016-01-26:
Can you remember on which channel you had the conversation? I can't find it in https://github.com/RIOT-OS/RIOT/pull/4287 and I'd like to clear this up soon because it seems to be causing quite a lot of confusion :(

cgundogan, 2016-01-26:
I think via the mailing list

authmillenon, 2016-01-26:
I think it was IRL, sadly. :-/ But if no one remembers by tonight I would say we just delete the deprecated application :-)

Lotterleben, 2016-01-26:
@OlegHahm voted to delete it here as well: https://lists.riot-os.org/pipermail/devel/2015-December/003261.html

authmillenon, 2016-01-26:
@haukepetersen can't remember any comments made on this issue, so just delete it ;-)

miri64, 2016-10-18:
Microcoap application in `applications` was removed so close

closed 2016-10-18 by miri64