RIOT-OS/RIOT #4684: USB BLE HCI protocol

a3zzat, 2016-01-25:
contributing to : "  USB Device Stack #3546 " 
RIOT don't have HCI for Bt\BLE so I will work on that.

[image]

jfischer-phytec-iot, 2016-01-26:
Hello a3zzat, I refactor #3890 and have added support for multiple interfaces and samr21 (samd21), the new version is not published, so let me know please before you start with the work.

a3zzat, 2016-01-27:
I will start working on implementing the HCI commands and events until you finish then we can link the code

phani2410, 2016-02-08:
I am interested to work on BLE Protocols. Can you please suggest me, what are the relevant things to be learnt for working on this issue.

PeterKietzmann, 2017-11-27:
@a3zzat, @phani2410 was there any progress since early 2016?

PeterKietzmann, 2018-06-26:
No progress since Feb. '16 nor feedback since Nov. '17. Closing this one. Please re-open when planning to take up work again

closed 2018-06-26 by PeterKietzmann