RIOT-OS/RIOT #4693: cpu/lpc2387: weird constructs and bit-rot

gebart, 2016-01-27:
The lpc2387 port has several problems in current master:

 - The i2c driver has its own subdirectory
 - Some documentation that should be on the wiki is located in `cpu/lpc2387/doc/lpc2387.doc` (a text file, not MS Word)
 - Missing a lot of Doxygen comments
 - Magic numbers in several implementation files
 - uses `F_CPU` macro
 - and so on...

miri64, 2016-10-18:
Ping?

miri64, 2019-08-10:
I think @maribu is working on this, right?

maribu, 2019-08-10:
Yep. But it will require a bunch of PRs to get it back up to standards. :-/

The first two however are ready for review

miri64, 2019-08-10:
Which PRs are those? It's always good to link issues and PRs ;)

miri64, 2020-02-11:
Ping @maribu @benpicco?

benpicco, 2020-02-11:
should be mostly resolved now.

maribu, 2020-02-11:
Yep. It is indeed getting in a pretty good shape. Once the still open PRs of @benpicco are in, this could be closed.

One final touch would be to check whether the use of thumb is beneficial. It would at least reduce programming time quite a bit due to the lower code size. Depending on how the memory is connected to the CPU, it can even increase performance (in case 16 bit thumb instructions can be fetched in one cycle, but 32 bit ARM instructions take two). But it seems only thumb version 1 is supported, so there might be indeed some trade off :-/

miri64, 2020-06-30:
> Once the still open PRs of @benpicco are in, this could be closed.

Are they in?

benpicco, 2020-06-30:
I don't have any more cleanup PRs open for lpc23xx

miri64, 2020-06-30:
Then let's close this finally!

closed 2020-06-30 by miri64