RIOT-OS/RIOT #4695: Makefiles: Move top level Makefile.* to its own directory?

gebart, 2016-01-27:
There are a lot of Makefile.* in the top level directory of the tree. Would it make sense to create a new directory for all kinds of additional makefile includes like Makefile.buildtest, Makefile.openocd etc?

Perhaps even renaming them to something shorter like `buildtest.mk`, `cortexm_common.mk` etc?

gebart, 2016-01-27:
I did a proof of concept in #4699, but let's keep the discussion thread about the idea in this issue.

gebart, 2016-03-07:
@authmillenon @OlegHahm @haukepetersen @kaspar030 @cgundogan and everyone else, do you agree with the idea? I don't want to keep #4699 open if the consensus is to keep the Makefiles as-is in the top level directory.

OlegHahm, 2016-03-07:
Sorry for the silence. I have no strong opinion yet, but for someone who works mostly with shell and vim an additional folder level is little bit disadvantageous and `makefiles/buildtest.mk` is also not shorter than `Makefile.buildtest`.

authmillenon, 2016-03-07:
I in fact would prefer to have them in a separate directory, *because* I'm a shell and vim user. 
Currently the root directory looks like this with `ls`:

```
boards           doc.txt               Makefile.base            Makefile.include        README.md            Vagrantfile
CONTRIBUTING.md  drivers               Makefile.buildtests      Makefile.modules        release-notes.txt
core             examples              Makefile.cflags          Makefile.pseudomodules  sys
cpu              LICENSE               Makefile.defaultmodules  Makefile.scan-build     tests
dist             Makefile              Makefile.dep             Makefile.vars           toolchain
doc              Makefile.application  Makefile.docker          pkg                     uncrustify-riot.cfg
```

With the proposed change they all those Makefile.* would would go to a subdirectory, removing a lot of clutter (ideally they shouldn't be edited as often anyway). Name-lengths aren't really an argument for me. For this tab-completion exists.

OlegHahm, 2016-03-07:
Adding a directory also adds one hit of the tab key. ;-)

> ideally they shouldn't be edited as often anyway

Well, I haven't thought about the question whether this is ideal or not, but in fact I open them quite often - at least some of them.

gebart, 2016-03-07:
My arguments are the same as @authmillenon. I guess it's a matter of personal taste but I prefer to have as few _files_ as possible in the top level directory to reduce clutter. The directory tree is already quite well organized and easy to navigate, at least the parts where I work often. Also, the number of Makefile.* in the top level has grown with the addition of new features such as docker and scan-build and may keep growing as more features appear.

gebart, 2016-03-07:
Another reason to restructure the files is to make it easier to understand for new developers. By organizing the makefiles as proposed in #4699 I was hoping to make it obvious where to look for if you want to add a new cool toolchain or add support for a new hardware debugger etc.

kaspar030, 2016-03-07:
+1

I would prefer to keep the makefiles in the directory structure, if possible, like cpu/coretexm_common.mk, and only put those that are currently in the root into makefiles/.

What do you think?

Kijewski, 2016-03-07:
I am in favor of moving all Makefile in $RIOTBASE (save Makefile.base and Makefile.include) into its own directory, because the top-directory should only contain the most important information. (If it wouldn't break so much, I'd even move Makefile.base.) Renaming the files into *.mk makes sense.

OlegHahm, 2016-03-07:
One advantage of having them in a dedicated folder would be that we could also add a README that explains their purpose.

cgundogan, 2016-03-09:
Another advantage of having them in a separate folder would be that we can have multiple build systems in parallel without convoluting our root folder

kaspar030, 2016-03-09:
I like! ;)

OlegHahm, 2016-03-10:
It seems we have a (rough) consensus.

gebart, 2016-03-22:
Reading through the discussion I reach the conclusion that the consensus is to arrange it along the lines of #4699

gebart, 2017-03-12:
After applying #4699 the makefiles tree looks like this:
```
makefiles
├── application.mk
├── arch/
│   ├── cortexm.mk
│   └── msp430.mk
├── bindist.mk
├── buildtests.mk
├── cflags.mk
├── defaultmodules.mk
├── docker.mk
├── features.mk
├── modules.mk
├── pseudomodules.mk
├── scan-build.mk
├── tests.mk
├── toolchain/
│   ├── gnu.mk
│   └── llvm.mk
├── tools/
│   ├── jlink.mk
│   ├── openocd.mk
│   └── serial.mk
└── vars.mk
```

gebart, 2017-04-05:
#4699 was merged.

closed 2017-04-05 by gebart