RIOT-OS/RIOT #4716: stm32f4: UART DMA causes lockup in xtimer_drift test

gebart, 2016-01-29:
Current tests/xtimer_drift in master does not run for me on stm32f4discovery, it hangs after trying to start the first slacker thread.
Disabling DMA in the UART TX by commenting out code seem to make the problem go away.

I suspect there is a race between the main thread using printf/puts right before the thread_create call and the newly created thread also using printf immediately after creation.

OlegHahm, 2016-01-29:
Related to https://github.com/RIOT-OS/RIOT/issues/4488?

gebart, 2016-01-30:
no, after looking into this a bit further it seems like the problem is that periph/uart is not thread safe, it is just that we have not seen any symptoms until DMA was implemented in a driver.

PR #4717 (2016-02-01): cpu/stm32f4: Fix DMA race bug (#4716)
Fixes #4716 by adding an implicit mutex for all UART writes from thread context.
closed 2016-02-01 by gebart