RIOT-OS/RIOT #4718: periph/uart: Add locking?

gebart, 2016-01-30:
Debugging #4716 led to some thoughts about the UART interface.
Should periph/uart follow the same design as the SPI and I2C drivers and introduce new functions for locking RX, TX before using from thread context?

Since UART is full duplex and asynchronous there would have to be two separate mutexes.

Example API:
```c
int uart_rx_acquire(uart_t dev);
int uart_rx_release(uart_t dev);
int uart_tx_acquire(uart_t dev);
int uart_tx_release(uart_t dev);
```

Locking would be implicit in stdio calls (high level API), but explicit if using uart_write (low level API).

My question is: Should we add locking as a part of the periph/uart API?

An alternative solution is to use implicit locking, like in #4717, but this would mean that periph/uart does not follow the same design as periph/spi and periph/i2c.

haukepetersen, 2016-02-05:
Just generalized this thought a little bit:  #4758

haukepetersen, 2016-02-05:
For this case, I am against the `acquire/release` for UART, as IMHO UART is not a shared resource per se, but rather multiplexed by layers above, so let the multiplexing layer take care of locking.

kaspar030, 2016-02-05:
> UART is not a shared resource per se, but rather multiplexed by layers above, so let the multiplexing layer take care of locking.

+1.

kaspar030, 2016-02-05:
An UART maps lovely on file interfaces (open, read, write, close, optionally with open(..., EXCLUSIVE for locking).

gebart, 2016-02-10:
I agree, an UART driver should not force the user to use any kind of locking

closed 2016-02-10 by gebart