RIOT-OS/RIOT #4723: net: gnrc: incorrect icmp reply checksum

kaspar030, 2016-01-31:
Somehow an echo reply has incorrect checksum.

Setup: <linux ethos> <-> <6lbr ethos> <6l0> <-> <6lr>

PCAP: `http://www.schleiser.de/~kaspar/incorrect-checksum.pcapng`

OlegHahm, 2016-01-31:
Can you upload to cloudshark?

kaspar030, 2016-01-31:
https://www.cloudshark.org/captures/51a3ef392796

OlegHahm, 2016-01-31:
So, this only happens when pinging IPv6 <-> 6LoWPAN (i.e. including a 6LBR)?

kaspar030, 2016-01-31:
Yes, both pinging between linux<->6lbr and 6lbr<->6lr is working fine.

OlegHahm, 2016-01-31:
Hm, ok, in this case I'm currently unable to test. Sorry.

kaspar030, 2016-01-31:
Sometimes it happens with just linux<->ethos:

https://www.cloudshark.org/captures/28ec6d114009

(linux as fe80::1, RIOT as fe80::2, on riot "ifconfig <n> add fe80::2, fibroute add :: via fe80::1 dev <n>". 
Unfortunately, this doesn't always happen...

gebart, 2016-01-31:
Maybe an uninitialized checksum variable or similar which just happens to
have the correct initial value most of the time?

kaspar030, 2016-02-04:
Ok, this one really tricked me.

Turned out that in ethos on the linux side, I didn't disable all character conversion done by the serial driver. That made ethos replace all incoming 0x0A with 0x0D. That didn't really cause problems, other than that the source address of my 2001:db8:... was replaced with 2001:ad8:..., which I totally didn't spot in the wireshark dumps...

(bug found, bug killed, closing issue)

closed 2016-02-04 by kaspar030

OlegHahm, 2016-02-04:
hehe, nice catch