RIOT-OS/RIOT #4724: gnrc: ipv6: link-local NDP problem?

kaspar030, 2016-01-31:
Hey,

If I add fe80::2 to RIOT ethernet (ethos) using "ifconfig <n> add fe80::2", then add fe80::1 to the ncache for that interface ("ncache add <n> fe80::1"), ipv6 outputs this:

```
main(): This is RIOT! (Version: 2016.03-devel-303-g2c5ec-booze-ethos_br_hack)
RIOT border router example application
All up, running the shell now
> ifconfig
ifconfig

Iface  5   HWaddr: 35:76  Channel: 26  Page: 0  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: 4
           Long HWaddr: 5a:55:44:43:99:18:b5:76
           AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  IPHC
           Source address length: 8
           Link type: wireless
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::5855:4443:9918:b576/64  scope: local
           inet6 addr: ff02::1:ff18:b576/128  scope: local [multicast]

Iface  6   HWaddr: 02:9e:ae:22:ee:fa
           MTU:1500  HL:64  RTR  RTR_ADV
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::9e:aeff:fe22:eefa/64  scope: local
           inet6 addr: ff02::1:ff22:eefa/128  scope: local [multicast]
           inet6 addr: ff02::2/128  scope: local [multicast]

> ifconfig 6 add fe80::2
ifconfig 6 add fe80::2

success: added fe80::2/64 to interface 6
> ncache add 6 feipv6: Router advertisement retransmission event received
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to zu (network byteorder 0018)
ipv6: set next header to 58
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 6
ethos send
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to zu (network byteorder 0018)
ipv6: set next header to 58
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 6
ethos send
ipv6: waiting for incoming message.
80::1
ncache add 6 fe80::1

success: added IPv6 address fe80::1 to neighbor cache
> ping6 fe80::1
ping6 fe80::1

ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to zu (network byteorder 000c)
ipv6: set next header to 58
ipv6: set packet source to fe80::2
ipv6: calculate checksum for upper header.
ipv6: add interface header to packet
ipv6: send unicast over interface 6
ipv6: waiting for incoming message.
ping timeout
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to zu (network byteorder 000c)
ipv6: set next header to 58
ipv6: set packet source to fe80::2
ipv6: calculate checksum for upper header.
ipv6: add interface header to packet
ipv6: send unicast over interface 6
ipv6: waiting for incoming message.
ping timeout
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to zu (network byteorder 000c)
ipv6: set next header to 58
ipv6: set packet source to fe80::2
ipv6: calculate checksum for upper header.
ipv6: add interface header to packet
ipv6: send unicast over interface 6
ipv6: waiting for incoming message.
ping timeout
--- fe80::1 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss
> 
```

The first two packets sent are router advertisements from RIOT, they arrive at the other end.
For the echo requests, RIOT says "ipv6: send unicast over interface 6", but that interface's send function is never called. I've added "ethos send" to the top of that function.

Now the question is, should RIOT try to solicit the MAC address of a neighbour cache entry on an interface when the entry doesn't have the l2 address set? (It seems it doesn't)

OlegHahm, 2016-01-31:
No, it shouldn't (and it didn't do so ast November, last time I checked this behavior). But if it doesn't send ut any NS, it still doesn't seem to do so. I advice to use a debugger to check where it exits.

kaspar030, 2016-01-31:
I think I solved my problem by not using the ncache for specifying an uplink, but just setting a route (fibroute add :: via fe80::1 dev <n>". That way RIOT correctly uses NDP to get fe80::1's l2 address.

I guess adding an l3 address without l2 address makes only sense with SLIP-

I'll close this for now.

closed 2016-01-31 by kaspar030

kaspar030, 2016-01-31:
(but if such an entry doesn't have any effect, the cli should mention that.)

OlegHahm, 2016-01-31:
Hm, yeah, printing a warning when trying to add a ncache entry for address-less link layers may make sense.