RIOT-OS/RIOT #4737: cortex-m: Hard fault after a thread exits

gebart, 2016-02-04:
... at least on these two Cortex-M4 platforms (tests/xtimer_msg_receive_timeout).

It seems like the function is trying to return to 0x00000000, which is bad...

stm32f4discovery:
```
2016-02-04 11:12:56,827 - INFO # �main(): This is RIOT! (Version: 2016.03-devel-298-gca8db-mazarin)
2016-02-04 11:12:57,828 - INFO # Message received: 44
2016-02-04 11:12:58,829 - INFO # Timeout!
2016-02-04 11:12:59,829 - INFO # Message received: 44
2016-02-04 11:13:00,830 - INFO # Timeout!
2016-02-04 11:13:01,832 - INFO # Message received: 44
2016-02-04 11:13:02,831 - INFO # Timeout!
2016-02-04 11:13:03,832 - INFO # Message received: 44
2016-02-04 11:13:04,832 - INFO # Timeout!
2016-02-04 11:13:05,833 - INFO # Message received: 44
2016-02-04 11:13:06,833 - INFO # Timeout!
2016-02-04 11:13:06,834 - INFO # 
2016-02-04 11:13:06,836 - INFO # Context before hardfault:
2016-02-04 11:13:06,838 - INFO #    r0: 0x2000015c
2016-02-04 11:13:06,839 - INFO #    r1: 0x40000000
2016-02-04 11:13:06,841 - INFO #    r2: 0x00000000
2016-02-04 11:13:06,842 - INFO #    r3: 0x2000084c
2016-02-04 11:13:06,844 - INFO #   r12: 0x00000000
2016-02-04 11:13:06,845 - INFO #    lr: 0x08000a81
2016-02-04 11:13:06,847 - INFO #    pc: 0x00000000
2016-02-04 11:13:06,848 - INFO #   psr: 0x6000002c
2016-02-04 11:13:06,852 - INFO # 
2016-02-04 11:13:06,853 - INFO # FSR/FAR:
2016-02-04 11:13:06,855 - INFO #  CFSR: 0x00020000
2016-02-04 11:13:06,856 - INFO #  HFSR: 0x40000000
2016-02-04 11:13:06,861 - INFO #  DFSR: 0x00000008
2016-02-04 11:13:06,862 - INFO #  AFSR: 0x00000000
2016-02-04 11:13:06,863 - INFO # Misc
2016-02-04 11:13:06,865 - INFO # EXC_RET: 0xfffffff1
2016-02-04 11:13:06,868 - INFO # Attempting to reconstruct state for debugging...
2016-02-04 11:13:06,869 - INFO # In GDB:
2016-02-04 11:13:06,870 - INFO #   set $pc=0x0
2016-02-04 11:13:06,871 - INFO #   frame 0
2016-02-04 11:13:06,872 - INFO #   bt
2016-02-04 11:13:06,872 - INFO # 
2016-02-04 11:13:06,878 - INFO # ISR stack overflowed by at least 72 bytes.
```

mulle:
```
2016-02-04 11:09:23,462 - INFO # main(): This is RIOT! (Version: 2016.03-devel-298-gca8db-mazarin)
2016-02-04 11:09:24,442 - INFO # Message received: 44
2016-02-04 11:09:25,421 - INFO # Timeout!
2016-02-04 11:09:26,400 - INFO # Message received: 44
2016-02-04 11:09:27,378 - INFO # Timeout!
2016-02-04 11:09:28,358 - INFO # Message received: 44
2016-02-04 11:09:29,337 - INFO # Timeout!
2016-02-04 11:09:30,315 - INFO # Message received: 44
2016-02-04 11:09:31,295 - INFO # Timeout!
2016-02-04 11:09:32,274 - INFO # Message received: 44
2016-02-04 11:09:33,253 - INFO # Timeout!
2016-02-04 11:09:33,254 - INFO # 
2016-02-04 11:09:33,255 - INFO # Context before hardfault:
2016-02-04 11:09:33,257 - INFO #    r0: 0x1fff0184
2016-02-04 11:09:33,259 - INFO #    r1: 0x1fff095c
2016-02-04 11:09:33,260 - INFO #    r2: 0x00000000
2016-02-04 11:09:33,262 - INFO #    r3: 0x1fff0884
2016-02-04 11:09:33,263 - INFO #   r12: 0x00000000
2016-02-04 11:09:33,265 - INFO #    lr: 0x00000c0d
2016-02-04 11:09:33,266 - INFO #    pc: 0x00000000
2016-02-04 11:09:33,268 - INFO #   psr: 0x60000055
2016-02-04 11:09:33,268 - INFO # 
2016-02-04 11:09:33,269 - INFO # FSR/FAR:
2016-02-04 11:09:33,270 - INFO #  CFSR: 0x00020000
2016-02-04 11:09:33,272 - INFO #  HFSR: 0x40000000
2016-02-04 11:09:33,273 - INFO #  DFSR: 0x00000000
2016-02-04 11:09:33,275 - INFO #  AFSR: 0x00000000
2016-02-04 11:09:33,276 - INFO # Misc
2016-02-04 11:09:33,277 - INFO # EXC_RET: 0xfffffff1
2016-02-04 11:09:33,281 - INFO # Attempting to reconstruct state for debugging...
2016-02-04 11:09:33,282 - INFO # In GDB:
2016-02-04 11:09:33,283 - INFO #   set $pc=0x0
2016-02-04 11:09:33,284 - INFO #   frame 0
2016-02-04 11:09:33,285 - INFO #   bt
2016-02-04 11:09:33,285 - INFO # 
2016-02-04 11:09:33,289 - INFO # ISR stack overflowed by at least 72 bytes.
```

gebart, 2016-02-04:
`tests/thread_exit` hard faults on stm32f4discovery, but works on mulle.

OlegHahm, 2016-02-04:
I re-delegate to someone with more experience on this board.

gebart, 2016-02-23:
#4883 should fix the `xtimer_msg_receive_timeout` problems, but the problem with `tests/thread_exit` on stm32f4discovery is unrelated.

gebart, 2016-02-23:
I think the stm32f4discovery problem may be related to newlib thread safety (printf maybe?) #4488

PeterKietzmann, 2016-06-28:
With *tests/thread_exit* I don't get a hard fault on the stm32f4discovery but it still behaves weird. Threads are not exiting, surprise surprise.

```
main(): This is RIOT! (Version: 2014.12-6535-g6050-kietzmann-Latitude-E6410-tests_xtimer_longterm_tickstousec)
main: starting
mnd: starting
2nd: starting
3th: starting
4th: starting
``` 
#4529 did not fix it

PeterKietzmann, 2017-01-16:
Hi again! tests/thread_exit seems to work correctly on stm324discovery with current master (at least for my setup). I don't have enough time/knowledge to dig deeper into the newlib/thread safe issue. Hope this result already helps?!

```
2017-01-16 16:13:10,754 - INFO # main: starting
2017-01-16 16:13:10,754 - INFO # main: exiting
2017-01-16 16:13:10,755 - INFO # 2nd: starting
2017-01-16 16:13:10,755 - INFO # 3rd: starting
2017-01-16 16:13:10,756 - INFO # 3rd: exiting
2017-01-16 16:13:10,756 - INFO # 2nd: exiting
2017-01-16 16:13:10,757 - INFO # 4th: starting
2017-01-16 16:13:10,757 - INFO # 4th: exiting
```

PeterKietzmann, 2017-01-18:
The issues described in this PR do not appear any more.

closed 2017-01-18 by PeterKietzmann