RIOT-OS/RIOT #4758: drivers/periph: some design considerations

haukepetersen, 2016-02-05:
We have lately started to put some efforts in re-modeling/overhauling our peripheral driver interfaces, and have already improved them quite a bit. Looking at all the open issues and PRs to this topic, I put some high-level thoughts on how we could find a generic concept which makes clear, which driver has to implement what kind of properties. I came to the following table (see explanation below):
```
Peripheral  Multiple    Shared  Atomic
adc         x           -       x (one shot)
adc         x           -       - (periodic)
cpuid       -           -       x
dac         x           -       -
gpio        x           -       -
i2c         x           x       x (master mode)
i2c         x           -       - (slave mode)
pwm         x           -       -
hrng        -           -       x
rtc         -           -       -
rtt         -           -       -
spi         x           x       x  (master mode)
spi         x           -       - (slave mode)
timer       x           -       -
uart        x           -       -

(x := yes, - := no)
```
**Multiple**: There is/can be more than one instance of this peripheral on a MCU. This means, that the interface must specify the used instance (what we do with the first parameter)
**Shared**: These peripherals are shared by 1-to-N 'consumers' (i.e. device drivers), typically bus systems. I don't consider `UART` to be shared, as it is an 1-to-1 connection (also for STDIO, only the syscall implementation uses the UART directly...).
**Atomic**: Must a peripheral only be active for the amount of time it is accessed or has it to be active the complete time? E.g. UART and timers must be active for RX and for running, respectively. But reading the CPUID, acquiring a single ADC value or sending some bytes via SPI can be considered as atomic in this sense.

Now let's see what implications these three groups bring us:
**Multiple**: Easy: all functions of an interface for these peripherals must include a `dev` parameter to identify the peripheral that should be used. Also we need means to configure the different instances of a peripheral on a per-board basis.

**Shared**: All shared devices must include means, to get and hand back exclusive access to a device. I think our current concept with `xx_aquire()` and `xx_release()` is the way to go here.

**Atomic**: This is the most interesting property, and I would propose to change our current thinking here slightly:
- all 'atomic' peripherals should be in off state per default
- so typically intially: enable device (turn on), initilize global options, disable device (turn off) 
- when used: turn on, do something, turn off
- this means: supply `poweron` and `poweroff` functions only for none-atomic peripherals, for all others call them implicit internally
- for **shared** peripherals the enabling and disabling should be put in their `aquire/release` functions

@gebart, @kaspar030, @thomaseichinger, @jfischer-phytec-iot, @cgundogan - opinions? 

If we all agree on this (or something else), I will write it down in more detail an put it into the `periph` driver doxygen documentation...

haukepetersen, 2016-02-05:
Oh, and forgot to write: i think all peripheral drivers need to be thread safe!

a3zzat, 2016-02-06:
so how are you going to translate that to a common APIs?

gebart, 2016-02-06:
I like the idea, though I have some general comments and concerns regarding the periph interfaces:

General: Should a thread expect that it may be preempted when calling a periph driver function?
From my view this is necessary in order to allow us to use implicit locking of hardware resources, e.g. the UART mutex in STM32F4 may cause a thread to yield whenever it does a printf call.

ADC: many MCUs have a limited number of actual ADC devices, but multiple channels, so two (application-) threads reading from different ADC channels may need to wait in turn to avoid getting their values messed up by the other thread. Maybe this should just be handled by implicit periph-driver-internal locking to make it thread safe?

SPI: I would like to (once again) propose to move chip select to be an integral part of the SPI drivers, not as separate GPIO calls. This is especially useful when implementing DMA for SPI transfers, if we want to save cpu cycles and power consumption.

Pin configuration: This has been discussed before, but is there any need for a pin configuration periph driver or should that be handled on a case-by-case basis per platform? (I have a PoC branch for a PORT driver for Kinetis which is shared between the periph drivers, but I am waiting for the driver API refactoring PRs to be merged before posting it)

haukepetersen, 2016-02-08:
good points:

> General:

I would say yes! As some peripheral accesses (like reading a random value, sampling an ADC channel, transmitting some bytes) can take a couple of moments, I think that threads have to expect to be preempted.

> ADC: 

This internal locking is exactly my point - I have already proposed the 'new' look of ADC drivers in #4430. This PR also introduces the means of ADC lines, making it very easy to scale on various number of ADC devices and their channels.

> SPI:

I completely agree, and I think this was also the consensus in #3216. I will PR my current WIP SPI branch as successor of #3216 today, so we can take the discussion on the details there.

> Pin configuration:

not sure what this pin configuration driver is. As I see it, the pin configuration should be handled by each peripheral driver itselft (as the periph drivers now best what there pins must be configured to). BUT they should of course re-use as much code as possible. So each CPU implementation should of course make sure to have internal functions that reduce code duplication of PIN configuration code. But this needs to be dealt with on a CPU basis, and therefore it does not make sense to have a RIOT wide pin configuration interface. Does this make sense and did I get your remark correctly?

Further I have started a table to get an overview on the current implementation status: https://docs.google.com/spreadsheets/d/16WWoQOkcwvMLQ_2QEJNoTwR1eglx2BpFUnZuWdbpWhg/edit?usp=sharing

This shows us, that there is quite a bit cleanup todo :-)

kaspar030, 2016-02-08:
> https://docs.google.com/spreadsheets/d/16WWoQOkcwvMLQ_2QEJNoTwR1eglx2BpFUnZuWdbpWhg/edit?usp=sharing

I fixed the "interface god". ;)

haukepetersen, 2016-02-08:
now you should be able to edit

haukepetersen, 2016-02-08:
@a3zzat: sorry, overlooked your question above... 

Translating this into an API is already basically done, our current periph interfaces do this mostly. The idea described above is rather to create a semi-formal classification scheme for periph drivers, that can be used to argue why e.g. certain interfaces have xx_poweron/off functions and others don't, or why some need to be acquired/released while other do not...

kaspar030, 2016-02-08:
If we touch all those APIs, I still think we should not mix interface and implementation.

```module_powerup(void);``` vs. ```module_powerup(module_t* handle)```

The overhead is negligable, the gain in flexibility is huge.

Already our little MCU's sometimes have different silicons, e.g., timer_A and timer_B with slightly different semantics (one 32bit, one 16bit). Currently the code path differences are implemented using if (TIMER_NUM == 0) {} else (...), already eating the one function pointer overhead.

SPI could be done in software, but not if spi_init(number) is hardcoded.
SPI with soft CS can be implemented as just a thin layer on top, once, for all boards, if the supplied pin's cannot be set by the SPI hardware (or it isn't wanted).

Multiple USB chips, ADC converters etc. could be connected to different buses. Not using pointer-based handles will prove very limiting.

Also, many interfaces might prove useful in upper layers. RTT or RTC could use the same interface, but with an intermediate drift compensation. Any 16bit timer can be made 32bit with a very thin layer.

The possibilities are endless, the overhead minimal, and it would probably pay off in the end by simplified implementations on many layers...

haukepetersen, 2016-02-08:
Hm, I have to say I don't understand your concept... Could you provide a concrete example, maybe for UART or similar?

kaspar030, 2016-02-08:
> Could you provide a concrete example, maybe for UART or similar?

Sure. Currently we have:

```
int uart_init(uart_t uart, uint32_t baudrate, uart_rx_cb_t rx_cb, void *arg);
void uart_write(uart_t uart, const uint8_t *data, size_t len);
```

I propose

```
int uart_init(uart_t *uart, uint32_t baudrate, uart_rx_cb_t rx_cb, void *arg);
void uart_write(uart_t *uart, const uint8_t *data, size_t len);
```
(uart parameter is a pointer)
with "uart_t" something like

```
typedef struct {
   int (*init)(uint32_t, ...);
   void(*write)(uart_t *, const uint8_t*, len);
} uart_t;
```   

like we do for netdev2.

haukepetersen, 2016-02-08:
I see, but I am not sure, if I like this much (yet?!). The main reason for non pointer device identifiers is their ease of use. It is very simple to statically define something like `driver_x uses UART_DEV(0)`, as `UART_DEV(0)` is constant. I fear that configuration of device drivers with non-constant peripheral identifiers get's out of hand. Also this concept does not make sense for all peripheral interfaces (e.g. GPIO), so we would loose some more consistency I guess.

Just to follow your thought: If I would use this scheme for the UART interface, my peripheral configuration in the CPU periph config would look somehing like this (i.e. for the `stm32f4`):
```c
typedef sturct {
    int (*init)(..);
    void (*write)(...);
    gpio_t rx_pin;
    gpio_t tx_pin;
    some_internal_cpu_specific_function_ptr; // e.g. for clock handling etc
    ...
} uart_config_t;
```
do I get this right? And wouldn't the overhead for saving these additional function pointers be quite a bit?

gebart, 2016-02-08:
I don't see the point of having the global functions uart_init etc if the struct itself contain pointers to those functions?
I do however see a benefit in using a pointer to a struct for containing hardware specific settings and states.

gebart, 2016-02-08:
@haukepetersen to save space you would make sure the driver struct resides in ROM and only the mutable values are placed in RAM (e.g. via a pointer).
It will require quite some work though to realize this object oriented interface, I believe.

kaspar030, 2016-02-08:
> Also this concept does not make sense for all peripheral interfaces (e.g. GPIO), so we would loose some more consistency I guess.

I think it does make sense even there. Think about the multiplexer example.

haukepetersen, 2016-02-08:
I have to say I am extremely skeptical for multiple reasons:
- I really fear the complexity - there are always more flexible and/or smaller concepts, but if you can not understand/use them very straight forward and easily, we gain nothing (take tinyOS as example). Also, one of the design objectives was to create an periph interface, that as easy to use as the Arduino shit (as in 'the *friendly* OS')
- use cases as e.g. co-existence soft- and hw-spi or usb-based and native UART are appealing, but to some extend we have to decide if they are really essential or if we can go without them
- as @gebart stated: changing this part does require significant work - are we ready to take this on? What I would VERY strongly oppose are two different versions of periph drivers like the situation we have with netdev/netdev2 at the moment...

Again, I am not opposing this concept, I am just being very careful...

haukepetersen, 2016-02-08:
I think to really get a feeling what the @kaspar030's proposed changes would look like, we probably would need an exemplary implmentation including the changed interface(s), at least one adapted board+CPU and at least one adapted device driver (best one that uses more than 1 peripheral interface). @kaspar030: are you up to provide this?

kaspar030, 2016-02-08:
> If I would use this scheme for the UART interface, my peripheral configuration in the CPU periph config would look somehing like this (i.e. for the stm32f4)

The uart_config_t can actually stay the same.

kaspar030, 2016-02-08:
> @kaspar030: are you up to provide this?

I would love to. Not sure I have time.

haukepetersen, 2016-02-08:
> I would love to. Not sure I have time.

That's exactly the problem. I think following the current periph paradigm, we would barely have a chance to get most of the implementations up-to-date in 1-2 release cycles, but deeper changes might stale this effort more and more... I just want to make sure we do something that we can actually get finished in some sense...

kaspar030, 2016-02-08:
I really don't think the pointer-based interface is more complex or hard to do... Will give it a shot.

haukepetersen, 2016-02-08:
if you say so - lets look at your example code then and try to clarify everything and decide based on that - everything that makes RIOT better should be tried :-)

kaspar030, 2016-02-08:
Well, the conversion was rather easy, but like this it's ~120b more code, which is a lot more than I thought.
I think a problem is defining all those const arrays within a header...

https://github.com/RIOT-OS/RIOT/compare/master...kaspar030:newperiph

haukepetersen, 2016-02-09:
to clarify: 120b just for the UART driver, right? In that state, I would say this is a no-go, so let's look for a more memory efficient solution...

haukepetersen, 2016-02-09:
and looking at the code, there is also a slight run-time penalty when needing to de-reference the `uart` struct first before calling the actual UART function, right?

Again, the periph interfaces are meant to abstract the actual peripheral hardware as slim and with as little overhead as possible, so every single bit of overhead should be well thought on and only spend if absolutely needed.

haukepetersen, 2016-02-09:
3rd remark: looking at the board configuration (periph_conf.h), I think the definition of `samd21_uart_devs` seems weird at that place. We should try to keep the board configuration as simple as possible, hopefully limited to defining pin mappings. But I think a solution could be, if we find a method to automatically create the `xx_uart_devs` array at some place (possibly by using #4290). Soft devices could also be 'linked' into that table automatically?!

gebart, 2016-02-10:
The discussion in #2528 is also relevant if redesigning periph driver concepts. I am in favour of moving all bus initialization outside the slave device drivers.

haukepetersen, 2016-02-10:
Actually, I think with the changes for I2C and SPI I envision the discussion in #2528 is solved: Let my finally PR my draft of the new SPI driver interface today, so I can demonstrate this...

haukepetersen, 2016-02-10:
done: #4780

miri64, 2016-10-18:
So #4780 and #4929 fix this?

gebart, 2016-10-18:
Only for spi and i2c. This issue is still relevant, but it might be beneficial to make the top post into a page in the doxygen documentation.

miri64, 2016-10-18:
Okay, I added a DocTF label

closed 2019-09-10 by stale[bot]