RIOT-OS/RIOT #4764: drivers: ethos needs OS X support

kaspar030, 2016-02-08:
Now that #4438 is in, the host side needs OS X support.
I don't have an OS X machine, so doing this remotely would suck.

Could anyone look into this? *wink* @thomaseichinger?

kYc0o, 2016-02-08:
I can test it in native MAC but I cannot succeed to flash a samr21... there is any special procedure? I have this tools installed and in my path:
```
193:gnrc_border_router facosta$ openocd --version
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00287-g85cec24 (2016-01-10-12:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
193:gnrc_border_router facosta$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

BytesGalore, 2016-02-08:
@kYc0o you need to use an OpenOCD version that has `cmsis-dap` and `hidapi-libusb` [1].
But you can easily build your own one just as we described for `watr.li`  \[2] (scroll down to `Building OpenOCD` part)

I hope this helps.

[1] https://github.com/RIOT-OS/RIOT/wiki/Board%3A-SAMR21-xpro#flashing-the-device
[2] http://watr.li/samr21-dev-setup-ubuntu.html

kYc0o, 2016-02-22:
I succeeded with the setup but I fairly understand how ethos is actually using tun/tap interfaces. I found that this library [1] can be very useful, but my question is: does ethos need a tun and a tap interface? or only a tap interface should be enough?

[1] https://github.com/LaKabane/libtuntap

kaspar030, 2016-02-22:
> does ethos need a tun and a tap interface? or only a tap interface should be enough?

Yes, tap only is enough, and ethos only uses a tap interface. You're probably confused by the fact both tun and tap are initialized using "/dev/net/tun", but the actual ioctl selects tap mode (using "IFF_TAP" parameter), which in ethos is supplied in the flag argument to "tun_alloc()".

kYc0o, 2016-02-22:
Ok great, yeah, I was talking with Oleg about that and I'll try with the tap interface. I'll try to create a PR with this functionality once #4725 is merged.

kaspar030, 2016-02-22:
> I'll try to create a PR with this functionality once #4725 is merged.

There might be a chicken and egg problem. Can't merge #4725 without ethos OS X support as it would break the border router example for OS X.
You could maybe create a PR on top of #4725, and we merge your's as soon as #4725 is in?

kYc0o, 2016-02-22:
Ok so I will prepare the PR as soon as possible, but since the implementation is quite different so far, the script could not be used on OS X (ip command does not exist...) so my idea is to give the parameters to another program which will set up the tap interface and then use it on ethos.

kaspar030, 2016-02-22:
> the script could not be used on OS X

You could rename the script to "start_network_linux.sh" and create another osx variant, possibly sourcing shared functionality.

kYc0o, 2016-03-23:
I think this cannot be done for this release, so I'll move it for the next one.

kYc0o, 2016-07-26:
And postponed again... :(

denizs, 2018-03-16:
Hey there - are there any news on this?

kYc0o, 2018-03-16:
I have a branch somewhere on which I didn't succeed to set up the routes properly... @denizs  do you have experience on setting up networking options (especially IPv6) on OS X?

miri64, 2018-03-17:
I guess the long term plan is to replace Ethos with either PPP or a multiplexed version of SLIP (e.g. [slipmux](https://tools.ietf.org/html/draft-bormann-t2trg-slipmux-02))

smlng, 2019-09-09:
no RIOT native on MacOS until RIOT native supports 64Bit, hence closing

miri64, 2019-09-09:
`ethos` is not run with `native`. So reasoning for closing not valid.

smlng, 2019-09-09:
> `ethos` is not run with `native`. So reasoning for closing not valid.

oops, I was a bit trigger happy there - still the whole tuntap routing is not working on macOS as it should (and does on Linux). I'll try to check again, but in the end either its working or broken beyond repair, hence this one can be closed.

closed 2020-04-12 by stale[bot]