RIOT-OS/RIOT #4788: pkg/micro-ecc: Makefile patch race

gebart, 2016-02-11:
There's a race condition in the Makefile which means that running multiple builds of tests/pkg_micro-ecc (e.g. Travis buildtest) will fail with "file already exists" errors or similar because the patch has been applied already.

I don't know how to fix this, but it is very annoying when Travis fails unrelated to the actual change..

OlegHahm, 2016-02-11:
https://github.com/RIOT-OS/RIOT/blob/master/pkg/micro-ecc/Makefile#L25 looks strange and unnecessarily complex to me.

haukepetersen, 2016-02-11:
+1 for being annoyed...

Kijewski, 2016-02-11:
Maybe use [flock](http://manpages.ubuntu.com/manpages/raring/man1/flock.1.html) like in [libfixmath/checkout.sh](https://github.com/RIOT-OS/RIOT/blob/60d44724199ee9ae9312f899bcf8fe4f22f39de5/pkg/libfixmath/checkout.sh)?

Kijewski, 2016-02-11:
Oops.

kaspar030, 2016-03-10:
I have a fix in #5018 (includes a commit that builds micro-ecc in BINDIR).

kaspar030, 2016-03-10:
#5018 is in. @gebart this should be fixed then?

gebart, 2016-03-11:
I guess so.
Does #5018 make the build system check out a fresh copy for every platform in make buildtest?

kaspar030, 2016-03-11:
Yes... Working on it.

miri64, 2016-10-18:
This should be fixed with #5018.

closed 2016-10-18 by miri64