RIOT-OS/RIOT #4789: ping6: timeout value too big?

OlegHahm, 2016-02-11:
In https://github.com/OlegHahm/RIOT/commit/c92d35e1de26c31ddd04d6361964994e5dfdd9f8 I decreased the timeout value to 1 second. However, I wonder if we should a) make the timeout configurable or b) decrease it even more (has anyone ever seen a ping on RIOT that had a RTT of more than 0.3s?).

**Pro** a) would be the most flexible solution. **Con** a) the ping6 command has already too many parameters IMO.
**Pro** b) would be a quiet convenient solution for most cases. **Con** b) may produce false negatives for really long routes or high latency links (e.g. cellular networks).

cgundogan, 2016-02-11:
I would opt for a configurable timeout **and** a reasonable adjustment to the default value (if parameter omitted)

cgundogan, 2016-02-11:
I am not sure if I ever saw pings with an rtt of > 300ms, but pinging multi-hop with several hops in between might reach that value?

cgundogan, 2016-03-23:
>I am not sure if I ever saw pings with an rtt of > 300ms, but pinging multi-hop with several hops in between might reach that value?

yesterday, I had rtts (ping6, multi-hop) of nearly 1s.

PeterKietzmann, 2019-07-23:
Ping timeout is configurable nowadays

closed 2019-07-23 by PeterKietzmann