RIOT-OS/RIOT #4794: doxygen showing wrong/confusing include directives

OlegHahm, 2016-02-11:
For instance, http://doc.riot-os.org/unionipv6__addr__t.html states that one has to include `<addr.h>` for using `ipv6_addr_t`, while in fact one has to write `include "net/ipv6/addr.h".

I have no idea how to influence this information in doxygen.

A-Paul, 2016-02-24:
Hi @OlegHahm I have found one *doxyfile-entry* `STRIP_FROM_INC_PATH` in charge for the representation of all generated *#define .** statements.
Once defined somehow, even with an empty string, all paths are absolute. It takes a list of strings which are matched lefthand with the path.

Regarding your example:
`STRIP_FROM_INC_PATH = /usr/local/src/RIOT/sys/include`
would strip
`#define </usr/local/src/RIOT/sys/include/net/ipv6/addr.h>`
to
`#define <net/ipv6/addr.h>`
as desired.

Before matching, environment variable expansion is performed on the strings, but no pathname expansion.

With `>$ export RIOTBASE=/usr/local/RIOT` before calling doxygen
`STRIP_FROM_INC_PATH = $(RIOTBASE)/sys/include`
would strip
`</usr/local/src/RIOT/sys/include/net/ipv6/addr.h>` to `<net/ipv6/addr.h>`
however
`$(RIOTBASE)/cpu/*/include`
wont match any of the *cpu* includes. They have to be provided separately. I don't think using
`$(RIOTBASE)/cpu` and getting e.g. `#define <x86/include/x86_interrupts.h>` is an option.

In essence a list of all possible include paths has to be generated, maybe on-the-fly by make, and assigned to `STRIP_FROM_INC_PATH`.
I think the effort is worth it.

OlegHahm, 2016-02-24:
> In essence a list of all possible include paths has to be generated, maybe on-the-fly by make, and assigned to STRIP_FROM_INC_PATH.
I think the effort is worth it.

I agree and I think a little shell script could do the trick here.

A-Paul, 2016-02-24:
Do you prefer building the list on every 'make doc*' run? Or putting it in the *riot.doxyfile* and update when include paths change?

OlegHahm, 2016-02-24:
In general putting in the doxyfile would be okay, but I fear that this list would get outdated in no time without any automatic updating mechanism. Building the list at build time would circumvent this problem.

A-Paul, 2016-02-24:
@OlegHahm, launched PR #4890 to solve this issue.

OlegHahm, 2016-04-12:
Fixed by #4890.

closed 2016-04-12 by OlegHahm