RIOT-OS/RIOT #4798: sys/crypto: the module is rather confusing user-unfriendly and should be refactored

BytesGalore, 2016-02-12:
While updating the doc regarding the `crypto` module (#4786 and #4787) we stumbled upon the module is rather confusing and its usage is very not user-friendly.

* One has to set an additional `CFLAG` in the used project `Makefile` to setup the right `cipher_context_t` buffer size
* The `return` values are not documented.
* The `cipher_context_t` seems also a bit confusing https://github.com/RIOT-OS/RIOT/pull/4787#issuecomment-183097964

basilfx, 2016-02-14:
I was looking for a way to override a cipher using a hardware, but it seems that [this commit](https://github.com/RIOT-OS/RIOT/pull/2759#issuecomment-92702756) (or similar) never made it.

Maybe something to also include?

BytesGalore, 2016-02-16:
@basilfx it sounds like a good idea to use weak symbols, I added a bulletpoint for it

OlegHahm, 2016-02-17:
Actually, I would vote against using compiler specific features like the weak pragma. It may introduce different behaviors dependent on the toolchain and version.

BytesGalore, 2016-02-17:
don't know, actually we use `__attribute__((weak))` in several places to exploit this behaviour, e.g. [1] [2] [3]

[1] https://github.com/RIOT-OS/RIOT/blob/master/sys/posix/pthread/pthread.c#L193
[2] https://github.com/RIOT-OS/RIOT/blob/master/sys/compat/vtimer/vtimer_compat.c#L84
[3] https://github.com/RIOT-OS/RIOT/blob/master/sys/oneway-malloc/oneway-malloc.c#L31

OlegHahm, 2016-02-17:
I know and I'm not happy about it, but I think vtimer_compat and oneway-malloc can be removed soon.

BytesGalore, 2016-02-17:
I mean I don't stick using pragmas, but until now I just found it a good "utility" for overloading functions and I never encountered that the behaviour was different with different configurations/compiler-versions.
But if you feel uncomfortable using pragmas I won't oppose against not use them :)

Kijewski, 2016-02-17:
Unless there is **any** evidence that `__attribute__((weak))` does not work as intended, I would use it as preferred way to enable overridable function(s|ality).

OlegHahm, 2016-02-17:
Even having [Wikipedia](https://en.wikipedia.org/wiki/Weak_symbol) recommending not to use it, is not evidence enough? 
"Weak symbols are not mentioned by C or C++ language standards; as such, inserting them into code is not very portable. Even if two platforms support the same or similar syntax for marking symbols as weak, the semantics may differ in subtle points, e.g. whether weak symbols during dynamic linking at runtime lose their semantics or not."
from http://www.sourceware.org/ml/libc-hacker/2000-06/msg00029.html

Kijewski, 2016-02-17:
> Even having Wikipedia recommending not to use it, is not evidence enough? 

IMO, no. 

> as such, inserting them into code is not very portable

IMO, moot.

> Even if two platforms support the same or similar syntax

The only problem would be native for Windows (does native run on windows?). Everything other platform uses ELF for its object files.

> dynamic linking at runtime

We don't do that.

OlegHahm, 2016-02-17:
> > dynamic linking at runtime

> We don't do that.

Not yet, but it's on the wishlist for long enough and I definitely hope we will have it soon. 

Moreover, we don't really need this pragma so far, so I oppose to "introduce" it here.

closed 2019-09-10 by stale[bot]