RIOT-OS/RIOT #4802: ping6: strange statitics

OlegHahm, 2016-02-12:
I just had a weird result while pinging between nodes:
```
2016-02-12 15:30:14,301 - INFO # 100 packets transmitted, 82 received, 18% packet loss, tim    e 32.0698928 s
2016-02-12 15:30:14,305 - INFO # rtt min/avg/max = 142.044/121.594/157.081 ms
```
I wonder how the average RTT can be smaller than the minimum. ;)

authmillenon, 2016-04-20:
Marked for next release.

authmillenon, 2016-04-20:
Could not reproduce, but the bug fixed in #5377 might be the cause of this.

kYc0o, 2016-07-11:
can anyone confirm if #5377 can be the source of this?

kYc0o, 2016-07-26:
Ok so let's see for the next release (again).

miri64, 2016-07-26:
I think that might have been solved. Lets close this and reopen in case something like this is seen again (it's a minor issue anyways).

closed 2016-07-26 by miri64