RIOT-OS/RIOT #4806: make: malformed archive when using parallelized building

basilfx, 2016-02-12:
As I mentioned in #4797, I sometimes get 'malformed archive' build errors.

I don't have enough knowledge on Makefiles or the build process, but I can trigger it with high probability with `make -jN` for `N > 1`, and it can be mitigated for `N = 1`.

Some things I noticed:
* If a build fails for `N > 1`, rebuilding using `N = 1` will also fail, because the archive is not removed.
* As far as I have seen, it always fails on the same archive file, e.g. `periph_cpuid/bin/stk3700/cortexm_common.a`.
* It fails for multiple tests/examples.

I did some runs for two applications that I could trigger the error: 
* Log file for `periph_adc` (three attempts): https://gist.github.com/basilfx/e6999e439258ef244b74
* Log file for `periph_cpuid` (with full `QUIET=0`): https://gist.github.com/basilfx/c6c12c9e68c23fffb6f3
* Log file for `periph_cpuid` (with different error): https://gist.github.com/basilfx/878e1b877afd9f9a7376 

For other boards I don't know if I have seen them. Possibly that it's only related to my EFM32 port, but it is based off the EZR32WG, and since `make clean && make -j1` work fine, I don't suspect it's in my port. Furthermore, my port doesn't modify any RIOT-OS files.

Details:
* MacBook Pro with quad core i7 (8 threads) and SSD
* OS X 10.11.2
* arm-gcc-embedded 5.2.1 20151202 (release)
* RIOT commit 9fcb031e2192aa51b113c5ce11071eba2becf904 with my EFM32 port.

gebart, 2016-02-12:
I believe that you may have added cortexm_common twice to the DIRS variable, which would potentially be able to cause these kinds of issues.

https://gist.github.com/basilfx/e6999e439258ef244b74,  make_j1_success.log:
```
...
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/basilfx/Desktop/RIOT/cpu/cortexm_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/basilfx/Desktop/RIOT/cpu/efm32_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/basilfx/Desktop/RIOT/cpu/cortexm_common
...
```

do a `git grep -E 'DIRS.*cortexm_common'` in your working copy to see if there are any redundant occurences. Typically the CPUs will have `DIRS = $(RIOTCPU)/cortexm_common` in their Makefile, but not the board, and not the "cpu common" Makefiles either.

basilfx, 2016-02-14:
Thanks for looking into it. I feel stupid for not seeing this :-)

I recompiled my 'testing suite' for the EFM32 and I did not notice any 'malformed archive' errors anymore.

closed 2016-02-14 by basilfx

gebart, 2016-02-14:
no problem, glad I could help!