RIOT-OS/RIOT #4811: msg: Propose to push messages down that task stack

DipSwitch, 2016-02-14:
In the `conn` interface, and some other module implementations. There are allot of `msg_send_to_self(&msg);` to push an message back to the msg queue if it was not for the current `msg_receive` callee. How about pushing the messages back through the task stack to calling functions / modules?

With something like this sort of psuedo code:
 
    /**
     * @brief POC for usage
     * @{
     */
    static bool _msg_callback(msg_t *m, void *arg) {
        if (m->type == MY_TYPE) {
            // handle and return true
            *(int*)arg = 1;
            return true;
        }

        return false;
    }

    static int main(int argc, char **argv) {
        volatile int state = 0;
        msg_receiver_t cb = { NULL, _msg_callback, &state };
        msg_register_receiver(&cb);

        msg_t m;
        for (;;) {
            if (msg_receive(&m) && !_msg_callback(&m)) {
                msg_send_self(&m, &cb);
            }

            switch (state) {
                case 0:
                // state process for processing data
                break;

                case 1:
                // do a conn transfer, request DNS, request TFTP or other
                // functions which are going over the network and can take
                // a long time to complete
                break;
            }
        }

        msg_unregister_receiver(&cb);
    }

    /**
     * @}
     */

    typedef struct msg_receiver {
        /**
         * @brief next msg_receiver in the chain
         */
        struct msg_receiver *next;

        /**
         * @brief callback to be called by called functions if
         *    they receive a msg that isn't for them
         *    if the msg is never accepted it will be dropped
         */
        bool (*msg_callback)(msg_t *msg, void *args);

        /**
         * @brief argument to pass, state object for example
         */
        void *arg;
    } msg_receiver_t;

    /**
     * @brief send self doesn't requeue, but searches for next msg handler
     */
    void msg_send_self(msg_t *m, msg_receiver_t *current) {
        while (current->next) {
            current = current->next;
            if (current->msg_callback(m, current->arg)) {
                return;
            }
        }

        // drop the message because it isn't important
    }

    /**
     * @brief before calling a function which handles messages you register your callback
     */
    void msg_register_receiver(msg_receiver_t *recv) {
        LL_PREPEND(current_thread->msg_receivers);
    }

    /**
     * @brief after returning from the function you unregister your callback
     */
    void msg_unregister_receiver(msg_receiver_t *recv) {
        // check if no upper frame forgot to unregister it's receiver
        assert(recv == current_thread->msg_receivers);

        current_thread->msg_receivers = recv->next;
    }

DipSwitch, 2016-02-23:
To cryptic? ;)

OlegHahm, 2016-02-23:
No, too lazy. ;)

DipSwitch, 2016-02-23:
Touche! :p

OlegHahm, 2016-02-23:
I don't get this switch-case statement over `state`.

closed 2019-09-10 by stale[bot]