RIOT-OS/RIOT #4822: kw2xrf: packet loss when packets get fragmented

cgundogan, 2016-02-15:
I realised that packets with big payloads get lost after they get fragmented by sixlowpan.
Reproduce:
Ping from  a `pba-d-01-kw2x` board to a `samr21-xpro` board:
`ping6 1 beef::1 59 0` will work
`ping6 1 beef::1 60 0` will fail (fragmented)

Surprisingly, enabling DEBUG output of sixlowpan "fixes" that error, so it seems to be a race or timing related thing.

PeterKietzmann, 2016-02-18:
If I see it correctly the send function is not guarded for ongoing transmissions (or other transactions). I **think** (as I don't really know the state machine) the correct way would be split the function (kw2xrf_set_sequence)[https://github.com/RIOT-OS/RIOT/blob/master/drivers/kw2xrf/kw2xrf.c#L136] into a "preparation" part to act as a guard and a separate execution function to trigger the actual transmission in case of making use of the pre-loading option. 

Placing `kw2xrf_set_sequence` (as done in #4844) or simply adding a delay at the beginning of the send function prevents from the described misbehaviour. That strengthens my above described assumption. @jfischer-phytec-iot could you please have a look at it as you know the internals of that device best? I will merge #4844 for now to fix the bug but would be nice to have an efficient solution.

jfischer-phytec-iot, 2016-07-28:
#5469 should fix it

miri64, 2016-10-18:
Can somebody confirm?

smlng, 2016-10-18:
@miri64: yes I can. I tested #5469 and besides its adoption of netdev2 it also helps to resolve this issue. However, @jfischer-phytec-iot wanted to work on the said PR a little more - I'd like to merge that one ASAP, though.

miri64, 2016-10-18:
Than discuss there please!

OlegHahm, 2017-01-16:
What's up here?

smlng, 2017-01-16:
close as soon as #5469 is merged

PeterKietzmann, 2017-01-24:
To me it seems #5469 does not fix this issue. @smlng please re-check.

PeterKietzmann, 2017-01-24:
I meant the follow-up PR #6453

cgundogan, 2017-06-07:
@smlng @PeterKietzmann what's the status on this?

smlng, 2017-06-07:
good question -care to check while you are running your own tests anyway 😉 ?

miri64, 2017-06-22:
A student of mine was able to reproduce it ;-)

smlng, 2017-06-22:
can you (or the student) elaborate a bit on: setup, test app, packet sizes and boards used - for the latter: obviously `pba-d-01-kw2x` but, which revision and/or interop with other boards or Linux?

miri64, 2017-06-22:
Interop with Raspberry Pi (as far as I know). How do you find out the revision? It's one of the ones where the `tcs37727` is already one sensor.

miri64, 2017-06-22:
@adrianghc is that about right?

smlng, 2017-06-22:
does it have 2x AA battery holders or that single flat cell one? The former should be the _newest_ (?) revision.

smlng, 2017-06-22:
Btw. I haven't observed the behavior when sending/receiving between two PhyNodes (`pba-d-01-kw2x`) using latest master (that is everything, after the radio driver (kw2xrf) rework to netdev2).

miri64, 2017-06-22:
flat cell.

adrianghc, 2017-06-22:
@miri64 Yes, but I am pretty sure it also failed with a samr-21-xpro instead of a Raspberry Pi in the past, though I haven't tested it with the current code yet.

smlng, 2017-06-22:
would be great, if you ( @adrianghc ) can test with other RIOT boards, i.e., samr21-xpro would be a good start. Linux/RasPi is a different story anyway ...

adrianghc, 2017-06-22:
I've just tested with the samr21-xpro with the same result.

miri64, 2017-06-26:
I just tested it with #5469 (which was said to fix the issue) as of the state on closing of that PR, but the same problem as on master exists. As soon as it reaches a certain size I don't receive a packet at the other end anymore (with ping6 that number is 92). I analyzed the problem from multiple sides:

1. Rebasing #5469 to maybe find the fix. Except for non-substential changes (`netdev2` naming, introduction of `luid` in master and a little different usage of the GPIO and SPI API [most likely to updates on that side]) there is no difference however.
2. Sniffing the traffic
3. Debug output in the driver (in combination with sniffing)

While (as already stated) (1) had proven to be fruitless, (2) and (3) revealed that, though fragmentation is initiated and the driver actually receives the packets, only the first fragment is send (resulting in it being dropped by the receiver, since no subsequent fragments are received for reassembly). The reason is [this line](https://github.com/RIOT-OS/RIOT/blob/9ce9dd601c7884980a84dccbc65a95c3a42e842a/drivers/kw2xrf/kw2xrf_netdev.c#L119).

miri64, 2017-06-26:
(Digging even deeper: `seq` is `XCVSEQ_TX_RX` at this point, which is why `kw2xrf_can_switch_to_idle()` returns 0).

miri64, 2017-06-26:
@smlng maybe you can do similar steps and find out, what your board is doing differently and why.

miri64, 2017-06-26:
@adrianghc a really shitty workaround would be to apply the following patch

```diff
diff --git a/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c b/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c
index d3c89a1..7d1fbcf 100644
--- a/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c
+++ b/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c
@@ -26,6 +26,7 @@
 #include "utlist.h"
 
 #include "rbuf.h"
+#include "xtimer.h"
 
 #define ENABLE_DEBUG    (0)
 #include "debug.h"
@@ -256,6 +257,7 @@ void gnrc_sixlowpan_frag_send(gnrc_sixlowpan_msg_frag_t *fragment_msg)
         thread_yield();
     }
     else {
+        xtimer_usleep(4317);
         /* (offset + (datagram_size - payload_len) < datagram_size) simplified */
         if (fragment_msg->offset < payload_len) {
             if ((res = _send_nth_fragment(iface, fragment_msg->pkt, payload_len, fragment_msg->datagram_size,
```

With this I was able to send out at least two fragments (with a significant increase of latency of course and risk of missing reception of course too), which is at max. 176 bytes and should be enough for your use-case. For more set a higher sleep period.

miri64, 2018-01-10:
#8304 only says it *partly* fixes this issue, so I re-open again, in case it still prevails (please test).

PeterKietzmann, 2018-01-10:
I said "partly" because pings with payloads greater than 900B (or so) did still not succeed. Maybe another issue which we need to figure out.

miri64, 2018-01-10:
> I said "partly" because pings with payloads greater than 900B (or so) did still not succeed. Maybe another issue which we need to figure out.

I can think of multiple reasons for that (apart from broken driver):

* Packet loss of 10% (900 B are already ~10 packets)
* Wrongly set MTU

miri64, 2018-01-10:
However, I agree that it is most likely that this hints at a totally different error class. So please open a new issue with your report regarding the remaining issues.

PR #8304 (2018-01-09): drivers/kw2xrf: finish ongoing transmission before sending next frame
### Contribution description

As described in below referenced issues, the KW2XRF driver currently does not wait for ongoing transmissions to finish before sending a next frame. If the device is busy, the second frame will simply be ignored. Problems usually occur with 6LoWPAN-fragmented  IPv6 packets. This PR provides a fix. Now `ping6` between a Phytec node an Atmel node with payloads up to 900. 

@gebart I added you as author because this is actually your solution from #7107. D'accord?

### Issues/PRs references

(Partly) [... cut]
closed 2018-01-10 by miri64