RIOT-OS/RIOT #4842: udp: checksum zero

immesys, 2016-02-17:
In the UDP checksum calculation, I think we might be missing the edge case where the checksum is zero.

Consider [this line](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/transport_layer/udp/gnrc_udp.c#L90) in gnrc_udp.c. That inverted value could be equal to zero (if csum was 0xFFFF). Then, [this line](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/transport_layer/udp/gnrc_udp.c#L250) is ambiguous because zero could mean a valid checksum or an invalid checksum.

I think the fix is to change _calc_checksum so that csum is only inverted if it is not equal to 0xFFFF. This is in line with section 8.1, bullet 4 of RFC 2460 that says that a valid, but zero checksum should be replaced with 0xFFFF.

I actually discovered this while trying to track down why I am seeing nodes sending UDP packets with the checksum field as zero, but that must be a different problem because I would expect gnrc_udp_calc_csum to return -ENOENT which should prevent the packet from being sent further up (I assume) so I still need to find out how the csum field is being sent as zero.

authmillenon, 2016-02-17:
I guess the most comprehensive solution would be to add a unittest to https://github.com/RIOT-OS/RIOT/blob/master/tests/unittests/tests-inet_csum/tests-inet_csum that triggers this corner case and see if it passes.

Yonezawa-T2, 2016-02-22:
If I understand it correctly, ICMPv6 (https://tools.ietf.org/html/rfc2463) keeps zero checksum as is and only UDP replaces zero checksum with 0xFFF, doesn't it?

Yonezawa-T2, 2016-02-22:
Created PR #4873.

immesys, 2016-02-22:
Many thanks

PR #4873 (2016-03-07): udp: fixed a corner case for checksum computation.
Fixes #4842.

RFC 2460 section 8.1 (https://tools.ietf.org/html/rfc2460#section-8.1)
states:
> if that
         computation yields a result of zero, it must be changed to hex
         FFFF for placement in the UDP header.  IPv6 receivers must
         discard UDP packets containing a zero checksum, and should log
         the error.

This patch fixes them.
This also adds test cases for `gnrc_udp_calc_csum`.
commit e92c355a (2016-03-07): udp: fixed a corner case for checksum computation.

Fixes #4842.

RFC 2460 section 8.1 (https://tools.ietf.org/html/rfc2460#section-8.1)
states:
> if that computation yields a result of zero, it must be changed to hex
> FFFF for placement in the UDP header.
closed 2016-03-07 by authmillenon