RIOT-OS/RIOT #4852: RFC: use continuous interface identifiers

OlegHahm, 2016-02-18:
The current solution using the driver's PID as an identifier for a network interface is a bit uninituitive and very dependent on the actual application. Since these PIDs are stored in an array anyway, using the array index for the identifier instead, would make the identifiers continuous and persistent over all applications on the same board (as long as we don't change the initialization code).

authmillenon, 2016-02-19:
If we do this only for the user representation this should be alright I guess. Stack-internally the PIDs should still be the identifier to prevent multiple lookups and to keep them consistent to netapi. But then the documentation needs to be *very* precise about this dualism. Otherwise it's getting confusing, too.

authmillenon, 2016-02-19:
(also: I would prefer hard-coded names, when it comes to intuitive (i.e. human readable) identifiers). lwIP uses 2 letters (set by driver) plus a digit (set by initialization order). This way we also don't have to "invent" names if we ever want to support POSIX's `<net/if.h>`)

OlegHahm, 2016-02-19:
Yes, I was talking only about the user/application interface - and I would stick to the number, especially since it comes without any overhead.

miri64, 2019-09-10:
This will become a must, if network interfaces are e.g. identified by pointers (e.g. with #9903)

miri64, 2020-06-30:
@leandrolanzieri @jia200x can you please look into this?

maribu, 2022-09-16:
IMO, this can be closed as wontfix. In the GNRC stack the advantage of using more human readable identifiers doesn't justify the overhead, and with lwip identifiers are already human readable.

If anyone disagrees, please reopen.

closed 2022-09-16 by maribu