RIOT-OS/RIOT #4855: 6LOWPAN ON SAMR21 Device

girisugu2, 2016-02-20:
Hi I am new to this,
 I am using two SAMR21-xpro boards.
I want to connect two boards automatically.For that i tried gnrc_networking.i tried rpl configurations to set parent and child nodes.But i don't know how to communicate over 6lowpan.please guide me.
Thanks in Advance.

With Regards,
R.Giritharan

cgundogan, 2016-02-28:
>But i don't know how to communicate over 6lowpan.please guide me.

What exactly do you mean by `communicate over 6lowpan`?
You can start a udp server on both nodes with `udp server start 8888` and then send udp packets to another node via `udp send <ip_addr> 8888 test`.  You can obtain the ip addresses via `ifconfig`.
The udp transmission will use 6lowpan, as this is the default for the transceiver you are using.
You can also test your connection by using the `ping` command. `ping <ip_addr>`. This will also use 6lowpan.

Could you make your question more specific with regard to what you want to transmit over 6lowpan?

girisugu2, 2016-02-28:
Hi I am new to RIOT.

                  I have two SAMR21-XPRO Devices.I am using RIOT OS to
 communicate between two board.I tried RPL Routing for mesh network.I set up
 one device is parent and another is child by RPL networking.I tried this
 link
 to set up RPL (https://github.com/RIOT-OS/RIOT/wiki/Tutorial%3A-RIOT-and-
 Multi-Hop-Routing-with-RPL).As per the above link all are working fine for
 me.
 My question is i want to communicate my parent node to wifi or ethernet
 based
 devices.For that which device i can use.I think boarder router is used to
 convert ipv6 to ipv4 or vice versa.So for boarder router which device
 shall i
 use to communicate with samr21.
 I am confusing over this.Please help me to find out the right solution.
 Thanks in advance.

 With Regards,
 R.Giritharan

authmillenon, 2016-02-28:
Sorry for sounding a little pissed, but it's neither helpful for any of us nor will it speed up "our process" if you just repeat the question over and over again. So please answer my question (and the questions posted to your other posts on both devel ML and here): What setup do you use? What pins did you connect your second UART to (which is needed for the process described in )? What prefix did you use? How did you configure the border router (you need to connect over /dev/ttyACMx for configuration)? And did you look into #4725, which simplifies a lot of bootstrapping and doesn't require you to open a second UART? 

> I think boarder router is used to convert ipv6 to ipv4 or vice versa.

There is currently no dual stacking in RIOT, we only have IPv6 support for now. The border router is just there to translate IPv6 over IEEE 802.15.4 (6LoWPAN) to IPv6 over X (Ethernet, SLIP, WiFi, you name it). If you don't have an IPv6 connection: There are tunnel brokers that  provide 6in4 tunneling or can find a number of other solutions using your own tunneling solution on Linux.

girisugu2, 2016-03-01:
ok ,i will explain my steps below,
1.>I tried boarder router as per the readme file in RIOT/examples/gnrc_boarder_router.
            As per the read me file everything works fine for me.(sudo ./tunslip6 affe::1/64 -t tun0 -s /dev/ttyUSB0)which opens the tunslip terminal.I can able to send messages from RIOT terminal to tunslip using txtsnd command.Even i can ping the affe::1 address which is configured in tunslip.

2.>And next is gnrc_networking.
           In this a.> i set the global IP adress in one board(root node).
                      b.>Then initializes the rpl init in root node.
                      c.>set the rpl init 2008::dead in root node i.e parent node
                      d.>After that in another node i.e child node i start the rpl init.
                      e.>So finally i create parent node and child node as per the rpl routing.

So finally,
    My Question is,
               I want to ping the child IP from linux PC i.e Outside world(for that only i use boarder router).So after that i want to communicate from outside world to child node via  parent node.

So how can i do that one?
how to push my data's to internet?
please help me...
                                            Thanks in advance!

With Regards,
R.Giritharan

cgundogan, 2016-03-01:
I am just wondering: did you also do `ifconfig 6 add affe::2` and `ncache add 6 affe::1` like described in the `README` for the border router example? Is the `2001:dead::` prefix propagated correctly, i.e. does the child node get a global ip address on its interface after configuring the `2001:dead::X` address on interface `5` of the border router?

BTW: we noticed that the current setup is not quite straight forward and we are working on an enhancement of the border router setup in #4725

girisugu2, 2016-03-01:
Thankyou for your reply,

I have only two samr21-xpro devices.
So i used that two devices for gnrc_networking for RPL setup(Parent node and child node).

So with this two devices how can i push the data's to internet (server).
I don't know how to do exactly.Here below i explain what i did exactly.

I want to control(send and receive) messages over internet via ethernet based devices or wifi.
So for now , what i did is,
i set up RPL (parent and child nodes)gnrc_networking.And after that i opened the tunslip in linux pc using SAMR21 UART.So i configure the global ip 2008::dead or something else in tunslip and RIOT terminal(parent node).
So my doubt is how to ping child node ip from linux pc using ping6 command.
what is the exact solution for my question?
Thanks in Advance!
With Regards,
R.Giritharan

cgundogan, 2016-03-01:
@girisugu2 you need to flash the `gnrc_border_router` example on one of the samr21-xpro boards. Currently, without modifying any Makefiles, you need to connect a second UART to your border router on PA23 (RX) and PA22 (TX). When you call `sudo ./tunslip6 affe::1/64 -t tun0 -s /dev/ttyUSB0`, then `/dev/ttyUSB0` needs to refer to this second UART that you connected additionally. After this setup, it's just following the [README](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_border_router).

Once #4725 is merged (hopefully today), this process will be simplified a lot and you won't need the second UART. So stay tuned!

girisugu2, 2016-03-01:
Thanks for your reply.
I tried this already.gnrc_boarder_router.
My Question is how to ping my child ip from linux pc...I am working on gnrc_networking for set up parent child nodes.

cgundogan, 2016-03-01:
are you able to ping from the parent to the child?

cgundogan, 2016-03-01:
Honestly, I do not understand your current setup:
Let's break it down:

You have two samr21-xpros.
One of them uses the `gnrc_border_router` example.
The other one uses the `gnrc_networking` example.

RPL is not included in the `gnrc_border_router` example by default. Did you add rpl to the Makefile?
BTW, your current setup with two nodes works completely without RPL. The prefix will be announced from the border router to the 6lowpan router via Prefix Information Options within the Router Advertisements (so, basically, automagically)

cgundogan, 2016-03-01:
Could you please show us the output of `ifconfig` on both of your nodes after setting up everything?

cgundogan, 2016-03-01:
And while we are at it, the output of `ncache` could also be interesting from both samr21-xpro boards.

girisugu2, 2016-03-01:
I here attach both board's log(BOARD A and BOARD B)
Both are configured gnrc_networking

BOARD A
RIOT/examples/gnrc_networking$ sudo make BOARD=samr21-xpro term
RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200"
2016-03-01 14:19:41,978 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
ifconfig
2016-03-01 14:20:38,888 - INFO # > ifconfig
2016-03-01 14:20:38,893 - INFO # Iface  7   HWaddr: 47:52  Channel: 26  Page: 0  NID: 0x23
2016-03-01 14:20:38,897 - INFO #            Long HWaddr: 5a:5a:41:6d:64:67:c7:52 
2016-03-01 14:20:38,904 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2016-03-01 14:20:38,910 - INFO #            AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  IPHC  
2016-03-01 14:20:38,913 - INFO #            Source address length: 8
2016-03-01 14:20:38,915 - INFO #            Link type: wireless
2016-03-01 14:20:38,921 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2016-03-01 14:20:38,926 - INFO #            inet6 addr: fe80::585a:416d:6467:c752/64  scope: local
2016-03-01 14:20:38,933 - INFO #            inet6 addr: ff02::1:ff67:c752/128  scope: local [multicast]
2016-03-01 14:20:38,934 - INFO #
set global ip
----------------
ifconfig 7 add 2001:db8::1
2016-03-01 14:21:20,012 - INFO # > ifconfig 7 add 2001:db8::1
2016-03-01 14:21:20,013 - INFO # success: added 2001:db8::1/64 to interface 7
Initializing RPL
-------------------
rpl init 7
2016-03-01 14:21:32,672 - INFO # > rpl init 7
2016-03-01 14:21:32,676 - INFO # successfully initialized RPL on interface 7
Configure RPL
--------------------
rpl root 1 2001:db8::1
2016-03-01 14:21:46,506 - INFO # > rpl root 1 2001:db8::1
2016-03-01 14:21:46,509 - INFO # successfully added a new RPL DODAG

RPL Result
----------------
rpl
2016-03-01 14:26:15,453 - INFO # rpl
2016-03-01 14:26:15,454 - INFO # instance table:	[X]	
2016-03-01 14:26:15,457 - INFO # parent table:	[ ]	[ ]	[ ]	
2016-03-01 14:26:15,457 - INFO # 
2016-03-01 14:26:15,461 - INFO # instance [1 | mop: 2 | ocp: 0 | mhri: 256 | mri 0]
2016-03-01 14:26:15,470 - INFO # 	dodag [2001:db8::1 | R: 256 | OP: Router | PIO: on | CL: his | TR(I=[0,8], k=20, c=10, TC=1s, TI=8s)]
ping6 fe80::5855:5064:9977:23ca(child IP i.e BOARD B)
2016-03-01 14:28:13,793 - INFO # ping6 fe80::5855:5064:9977:23ca
2016-03-01 14:28:13,808 - INFO # 12 bytes from fe80::5855:5064:9977:23ca: id=83 seq=1 hop limit=64 time = 8.223 ms
2016-03-01 14:28:14,827 - INFO # 12 bytes from fe80::5855:5064:9977:23ca: id=83 seq=2 hop limit=64 time = 8.855 ms
2016-03-01 14:28:15,843 - INFO # 12 bytes from fe80::5855:5064:9977:23ca: id=83 seq=3 hop limit=64 time = 6.944 ms
2016-03-01 14:28:15,847 - INFO # --- fe80::5855:5064:9977:23ca ping statistics ---
2016-03-01 14:28:15,853 - INFO # 3 packets transmitted, 3 received, 0% packet loss, time 2.0646154 s
2016-03-01 14:28:15,857 - INFO # rtt min/avg/max = 6.944/8.007/8.855 ms

BOARD B
--------------
RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM1" -b "115200"
2016-03-01 14:20:28,564 - INFO # Connect to serial port /dev/ttyACM1
Welcome to pyterm!
Type '/exit' to exit.
ifconfig
2016-03-01 14:20:33,384 - INFO # ifconfig
2016-03-01 14:20:33,393 - INFO # Iface  7   HWaddr: 23:ca  Channel: 26  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: hu 
2016-03-01 14:20:33,397 - INFO #            Long HWaddr: 5a:55:50:64:99:77:23:ca 
2016-03-01 14:20:33,402 - INFO #            AUTOACK  CSMA  MTU:1280  HL:hu  6LO  RTR  IPHC  
2016-03-01 14:20:33,405 - INFO #            Source address length: 8
2016-03-01 14:20:33,408 - INFO #            Link type: wireless
2016-03-01 14:20:33,413 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 14:20:33,419 - INFO #            inet6 addr: fe80::5855:5064:9977:23ca/hu  scope: local
2016-03-01 14:20:33,425 - INFO #            inet6 addr: ff02::1:ff77:23ca/hu  scope: local [multicast]
2016-03-01 14:20:33,426 - INFO #            
rpl init 7
2016-03-01 14:22:01,224 - INFO # > rpl init 7
2016-03-01 14:22:01,228 - INFO # successfully initialized RPL on interface 7
rpl
2016-03-01 14:22:04,742 - INFO # > rpl
2016-03-01 14:22:04,744 - INFO # instance table:	[X]	
2016-03-01 14:22:04,746 - INFO # parent table:	[X]	[ ]	[ ]	
2016-03-01 14:22:04,747 - INFO # 
2016-03-01 14:22:04,751 - INFO # instance [1 | mop: 2 | ocp: 0 | mhri: 256 | mri 0]
2016-03-01 14:22:04,759 - INFO # 	dodag [2001:db8::1 | R: 512 | OP: Router | CL: his | TR(I=[0,8], k=20, c=10, TC=lus, TI=lus)]
2016-03-01 14:22:04,766 - INFO # 		parent [addr: fe80::585a:416d:6467:c752 | rank: 256 | lifetime: 119s]
ping6 fe80::585a:416d:6467:c752(parent IP i.e BOARD A)
2016-03-01 14:27:23,801 - INFO # > ping6 fe80::585a:416d:6467:c752
2016-03-01 14:27:23,817 - INFO # 12 bytes from fe80::585a:416d:6467:c752: id=83 seq=1 hop limit=hu time = 64.008 ms
2016-03-01 14:27:24,834 - INFO # 12 bytes from fe80::585a:416d:6467:c752: id=83 seq=2 hop limit=hu time = 64.008 ms
2016-03-01 14:27:25,851 - INFO # 12 bytes from fe80::585a:416d:6467:c752: id=83 seq=3 hop limit=hu time = 64.008 ms
2016-03-01 14:27:25,855 - INFO # --- fe80::585a:416d:6467:c752 ping statistics ---
2016-03-01 14:27:25,861 - INFO # 3 packets transmitted, 3 received, 0% packet loss, time 2.0648265 s
2016-03-01 14:27:25,864 - INFO # rtt min/avg/max = 8.227/8.650/8.866 ms

cgundogan, 2016-03-01:
>Both are configured gnrc_networking

This does not work. One of the samr21-xpro boards needs to be flashed with the `gnrc_border_router` example in order to achieve what you desire.

girisugu2, 2016-03-01:
ok Fine.But if i use one as gnrc_boarder_router how can i set up as parent and child nodes.

girisugu2, 2016-03-01:
So no need RPL?

cgundogan, 2016-03-01:
You can omit RPL for your use case with two samr21-xpros. If you really need multi-hop on the lowpan side, then you can include the rpl module in the Makefile of the `gnrc_border_router` example, but AFAIK, this was never tested. The boder router might get confused with the additional fib entries of RPL.

cgundogan, 2016-03-01:
>The boder router might get confused with the additional fib entries of RPL.

@OlegHahm do you have something smart to say for this matter? (:

OlegHahm, 2016-03-01:
I'm not smart.

OlegHahm, 2016-03-01:
But in general, I see no problem with RPL on the 6LBR and I think @kYc0o have even tried.

girisugu2, 2016-03-01:
ok let me explain clearly,

          IOT(INTERNET OF THINGS) means connect devices with internet.For that my project is home automation..So i have different samr21 boards(now for testing i have two samr21).
I want to control that two boards via internet.
For that i tried RIOT .So i tried gnrc_networking and gnrc_border_router examples.As per the reame files from border_router and gnrc_networking works perfectly for me.

I think upto this everything is ok.
So after that i want to communicate the SAMR21 boards to internet.So only for one device i  think gnrc_networking is ok (to connect multiple nodes).
And to connect Parent node to IOT server gnrc_border_router is the only choice i think.
The border_router converts radio device datas to  normal internet .

This is my requirement.
So please tell me frankly about my ideas and your suggestions.

Thanks in advance
With Regards,
R.Giritharan

girisugu2, 2016-03-01:
I am wrong means please help how to do?

cgundogan, 2016-03-01:
@girisugu2 can you show me again your `ìfconfig` and `ncache` output of both nodes? But this time one of the nodes **must** the `gnrc_border_router` example (with all steps included that are outlined in the REAME of the border router example). The other node can be flashed with the `gnrc_networking` example.

girisugu2, 2016-03-01:
this is the gnrc_networking example...
how to configure ncache.
what is the use of ncache in RIOT
2016-03-01 15:18:29,063 - INFO # > ifconfig
2016-03-01 15:18:29,069 - INFO # Iface  7   HWaddr: 47:52  Channel: 26  Page: 0  NID: 0x23
2016-03-01 15:18:29,073 - INFO #            Long HWaddr: 5a:5a:41:6d:64:67:c7:52 
2016-03-01 15:18:29,080 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2016-03-01 15:18:29,085 - INFO #            AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  IPHC  
2016-03-01 15:18:29,088 - INFO #            Source address length: 8
2016-03-01 15:18:29,091 - INFO #            Link type: wireless
2016-03-01 15:18:29,096 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2016-03-01 15:18:29,102 - INFO #            inet6 addr: fe80::585a:416d:6467:c752/64  scope: local
2016-03-01 15:18:29,108 - INFO #            inet6 addr: ff02::1:ff67:c752/128  scope: local [multicast]
2016-03-01 15:18:29,109 - INFO # 
ncache
2016-03-01 15:19:47,607 - INFO # ncache
2016-03-01 15:19:47,613 - INFO # IPv6 address                    if  L2 address                state       type
2016-03-01 15:19:47,620 - INFO # ------------------------------------------------------------------------------

girisugu2, 2016-03-01:
One more doubt,
I i run gnrc_networking in one board,
gnrc_boarder_router in another means how to communicate between these two boards(gnrc_border_router to gnrc_networking).
Think in border router there is no UDP

cgundogan, 2016-03-01:
the more interesting part is the `ifconfig` and `ncache` output of the other node (; Did you connect your border router?

cgundogan, 2016-03-01:
>Think in border router there is no UDP

we can talk about this once we connect your border router correctly to the linux machine

girisugu2, 2016-03-01:
no i didn't connect.
Ok what shall i do?
please give me the step by step procedures.
I will follow your guidence.
Now i will flash one board as gnrc_networking and another as gnrc_border_router ok

cgundogan, 2016-03-01:
correct. Do you have a second USB TTL serial cable at hand?

girisugu2, 2016-03-01:
yes i have..In thai one i want to open tunslip right?

cgundogan, 2016-03-01:
Did you connect your second UART? Pins on the border router: PA23 (RX) and PA22 (TX) ?

girisugu2, 2016-03-01:
this is tunslip log,which is opened second uart
RIOT/dist/tools/tunslip$ sudo ./tunslip6 affe::1/64 -t tun0 -s /dev/ttyUSB0
[sudo] password for oviya: 
********SLIP started on ``/dev/ttyUSB0''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` up
ifconfig tun0 add affe::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:127.0.1.1  P-t-P:127.0.1.1  Mask:255.255.255.255
          inet6 addr: fe80::1/64 Scope:Link
          inet6 addr: affe::1/64 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

girisugu2, 2016-03-01:
so whuch ip i need to set up as a global ip in RIOT terminal of gnrc_border_router

cgundogan, 2016-03-01:
>so whuch ip i need to set up as a global ip in RIOT terminal of gnrc_border_router

`ifconfig 6 add affe::2`
`ncache add 6 affe::1`

cgundogan, 2016-03-01:
after this please verfiy if you can ping from linux `affe::2` and also verify if you can ping from the border router your linux machine `affe::1`

girisugu2, 2016-03-01:
LOG For BOARD B(border_router)

ifconfig
2016-03-01 15:29:22,091 - INFO # ifconfig
2016-03-01 15:29:22,100 - INFO # Iface  5   HWaddr: 23:ca  Channel: 26  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: hu 
2016-03-01 15:29:22,104 - INFO #            Long HWaddr: 5a:55:50:64:99:77:23:ca 
2016-03-01 15:29:22,110 - INFO #            AUTOACK  CSMA  MTU:1280  HL:hu  6LO  RTR  RTR_ADV  IPHC  
2016-03-01 15:29:22,113 - INFO #            Source address length: 8
2016-03-01 15:29:22,116 - INFO #            Link type: wireless
2016-03-01 15:29:22,121 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 15:29:22,127 - INFO #            inet6 addr: fe80::5855:5064:9977:23ca/hu  scope: local
2016-03-01 15:29:22,133 - INFO #            inet6 addr: ff02::1:ff77:23ca/hu  scope: local [multicast]
2016-03-01 15:29:22,138 - INFO #            inet6 addr: ff02::2/hu  scope: local [multicast]
2016-03-01 15:29:22,139 - INFO #            
2016-03-01 15:29:22,140 - INFO # Iface  6  
2016-03-01 15:29:22,144 - INFO #            MTU:1280  HL:hu  RTR  RTR_ADV  
2016-03-01 15:29:22,146 - INFO #            Link type: wired
2016-03-01 15:29:22,152 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 15:29:22,157 - INFO #            inet6 addr: ff02::2/hu  scope: local [multicast]
2016-03-01 15:29:22,158 - INFO #            
ifconfig 6 add affe::2
2016-03-01 15:38:14,233 - INFO # > ifconfig 6 add affe::2
2016-03-01 15:38:14,237 - INFO # success: added affe::2/64 to interface 6
ncache add 6 affe::1
2016-03-01 15:38:20,729 - INFO # > ncache add 6 affe::1
2016-03-01 15:38:20,734 - INFO # success: added IPv6 address affe::1 to neighbor cache
ping6 affe::1
2016-03-01 15:39:05,712 - INFO # > ping6 affe::1
2016-03-01 15:39:05,728 - INFO # 12 bytes from affe::1: id=83 seq=1 hop limit=hu time = 64.010 ms
2016-03-01 15:39:06,745 - INFO # 12 bytes from affe::1: id=83 seq=2 hop limit=hu time = 64.010 ms
2016-03-01 15:39:07,763 - INFO # 12 bytes from affe::1: id=83 seq=3 hop limit=hu time = 64.010 ms
2016-03-01 15:39:07,766 - INFO # --- affe::1 ping statistics ---
2016-03-01 15:39:07,772 - INFO # 3 packets transmitted, 3 received, 0% packet loss, time 2.0649218 s
2016-03-01 15:39:07,775 - INFO # rtt min/avg/max = 10.471/10.573/10.727 ms

Linux PC Log
-----------------
/RIOT/dist/tools/tunslip$ ping6 affe::2
PING affe::2(affe::2) 56 data bytes
64 bytes from affe::2: icmp_seq=1 ttl=64 time=19.8 ms
64 bytes from affe::2: icmp_seq=2 ttl=64 time=19.1 ms
^C
--- affe::2 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 19.138/19.506/19.875/0.394 ms

cgundogan, 2016-03-01:
great! the first step is done. Now, still on the border router, write: `ifconfig 5 add affe::3`.
After this step, look at the `ifconfig` output of the other node, which runs `gnrc_networking`.

girisugu2, 2016-03-01:
Hi i am extreamly sorry.Power went off .Just now it came.

girisugu2, 2016-03-01:
shall we continue now

cgundogan, 2016-03-01:
sure, can you show me the output of `ifconfig` of the second node (**not** the border router) after doing the same steps as above?

girisugu2, 2016-03-01:
LOG OF GNRC_BORDER_ROUTER

RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM1" -b "115200"
2016-03-01 18:51:22,185 - INFO # Connect to serial port /dev/ttyACM1
Welcome to pyterm!
Type '/exit' to exit.
ifconfig
2016-03-01 18:51:27,217 - INFO # ifconfig
2016-03-01 18:51:27,226 - INFO # Iface  5   HWaddr: 23:ca  Channel: 26  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: hu 
2016-03-01 18:51:27,230 - INFO #            Long HWaddr: 5a:55:50:64:99:77:23:ca 
2016-03-01 18:51:27,236 - INFO #            AUTOACK  CSMA  MTU:1280  HL:hu  6LO  RTR  RTR_ADV  IPHC  
2016-03-01 18:51:27,239 - INFO #            Source address length: 8
2016-03-01 18:51:27,242 - INFO #            Link type: wireless
2016-03-01 18:51:27,247 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 18:51:27,253 - INFO #            inet6 addr: fe80::5855:5064:9977:23ca/hu  scope: local
2016-03-01 18:51:27,259 - INFO #            inet6 addr: ff02::1:ff77:23ca/hu  scope: local [multicast]
2016-03-01 18:51:27,264 - INFO #            inet6 addr: ff02::2/hu  scope: local [multicast]
2016-03-01 18:51:27,265 - INFO #            
2016-03-01 18:51:27,266 - INFO # Iface  6  
2016-03-01 18:51:27,270 - INFO #            MTU:1280  HL:hu  RTR  RTR_ADV  
2016-03-01 18:51:27,272 - INFO #            Link type: wired
2016-03-01 18:51:27,278 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 18:51:27,283 - INFO #            inet6 addr: ff02::2/hu  scope: local [multicast]
2016-03-01 18:51:27,284 - INFO #            
ifconfig 5 add affe::3
2016-03-01 18:51:49,067 - INFO # > ifconfig 5 add affe::3
2016-03-01 18:51:49,072 - INFO # success: added affe::3/64 to interface 5
ifconfig 6 add affe::2
2016-03-01 18:52:01,955 - INFO # > ifconfig 6 add affe::2
2016-03-01 18:52:01,959 - INFO # success: added affe::2/64 to interface 6
ncache add 6 affe::1
2016-03-01 18:53:10,298 - INFO # > ncache add 6 affe::1
2016-03-01 18:53:10,303 - INFO # success: added IPv6 address affe::1 to neighbor cache

girisugu2, 2016-03-01:
Log for gnrc_networking
---------------------------------
ifconfig
2016-03-01 18:54:32,999 - INFO # > ifconfig
2016-03-01 18:54:33,008 - INFO # Iface  7   HWaddr: 47:52  Channel: 26  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: hu 
2016-03-01 18:54:33,013 - INFO #            Long HWaddr: 5a:5a:41:6d:64:67:c7:52 
2016-03-01 18:54:33,018 - INFO #            AUTOACK  CSMA  MTU:1280  HL:hu  6LO  RTR  IPHC  
2016-03-01 18:54:33,021 - INFO #            Source address length: 8
2016-03-01 18:54:33,024 - INFO #            Link type: wireless
2016-03-01 18:54:33,029 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 18:54:33,035 - INFO #            inet6 addr: fe80::585a:416d:6467:c752/hu  scope: local
2016-03-01 18:54:33,041 - INFO #            inet6 addr: ff02::1:ff67:c752/hu  scope: local [multicast]
2016-03-01 18:54:33,042 - INFO #

cgundogan, 2016-03-01:
OK, the second node did not pick up the prefix from the border router. Could you just reset the second node (**not** the border router) by pressing the reset button and then show me again the `ifconfig` output of that node?

girisugu2, 2016-03-01:
2016-03-01 18:57:03,911 - INFO # > main(): This is RIOT! (Version: 2015.12-devel-711-g5622-oviya-Tech)
2016-03-01 18:57:03,915 - INFO # RIOT network stack example application
2016-03-01 18:57:03,917 - INFO # All up, running the shell now
ifconfig
2016-03-01 18:57:08,086 - INFO # > ifconfig
2016-03-01 18:57:08,095 - INFO # Iface  7   HWaddr: 47:52  Channel: 26  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: hu 
2016-03-01 18:57:08,099 - INFO #            Long HWaddr: 5a:5a:41:6d:64:67:c7:52 
2016-03-01 18:57:08,105 - INFO #            AUTOACK  CSMA  MTU:1280  HL:hu  6LO  RTR  IPHC  
2016-03-01 18:57:08,108 - INFO #            Source address length: 8
2016-03-01 18:57:08,111 - INFO #            Link type: wireless
2016-03-01 18:57:08,116 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 18:57:08,122 - INFO #            inet6 addr: fe80::585a:416d:6467:c752/hu  scope: local
2016-03-01 18:57:08,128 - INFO #            inet6 addr: ff02::1:ff67:c752/hu  scope: local [multicast]
2016-03-01 18:57:08,129 - INFO #

girisugu2, 2016-03-01:
Done!

cgundogan, 2016-03-01:
What *should* happen is, that your second node picks up the `affe::/64` prefix and generates a global unicast IPv6 address on interface 7. However, your output shows that there is no global IPv6 address starting with `affe::`. Can you try to do `ping6 ff02::1` from the second node and verify that the border router answeres?

cgundogan, 2016-03-01:
@girisugu2 I think you are not using the current master. Could you please rebase your branch onto the current master?

girisugu2, 2016-03-01:
means?

girisugu2, 2016-03-01:
log for gnrc_networking

ifconfig
2016-03-01 19:07:31,178 - INFO # > ifconfig
2016-03-01 19:07:31,187 - INFO # Iface  7   HWaddr: 47:52  Channel: 26  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: hu 
2016-03-01 19:07:31,191 - INFO #            Long HWaddr: 5a:5a:41:6d:64:67:c7:52 
2016-03-01 19:07:31,197 - INFO #            AUTOACK  CSMA  MTU:1280  HL:hu  6LO  RTR  RTR_ADV  IPHC  
2016-03-01 19:07:31,200 - INFO #            Source address length: 8
2016-03-01 19:07:31,203 - INFO #            Link type: wireless
2016-03-01 19:07:31,208 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2016-03-01 19:07:31,214 - INFO #            inet6 addr: fe80::585a:416d:6467:c752/hu  scope: local
2016-03-01 19:07:31,220 - INFO #            inet6 addr: ff02::1:ff67:c752/hu  scope: local [multicast]
2016-03-01 19:07:31,225 - INFO #            inet6 addr: ff02::2/hu  scope: local [multicast]
2016-03-01 19:07:31,231 - INFO #            inet6 addr: affe::585a:416d:6467:c752/hu  scope: global
2016-03-01 19:07:31,232 - INFO #            
ping6 ff02::1
2016-03-01 19:07:33,299 - INFO # > ping6 ff02::1
2016-03-01 19:07:33,315 - INFO # 12 bytes from fe80::5855:5064:9977:23ca: id=85 seq=1 hop limit=hu time = 64.008 ms
2016-03-01 19:07:34,342 - INFO # 12 bytes from fe80::5855:5064:9977:23ca: id=85 seq=2 hop limit=hu time = 64.017 ms
2016-03-01 19:07:35,361 - INFO # 12 bytes from fe80::5855:5064:9977:23ca: id=85 seq=3 hop limit=hu time = 64.008 ms
2016-03-01 19:07:35,363 - INFO # --- ff02::1 ping statistics ---
2016-03-01 19:07:35,369 - INFO # 3 packets transmitted, 3 received, 0% packet loss, time 2.0656639 s
2016-03-01 19:07:35,373 - INFO # rtt min/avg/max = 8.590/11.450/17.169 ms

girisugu2, 2016-03-01:
is there any configurations i need to do in gnrc_networking node???

cgundogan, 2016-03-01:
ok you got a global ipv6 address now for the second node. Your IPv6 address is `affe::585a:416d:6467:c752`.
Try to ping that from linux

girisugu2, 2016-03-01:
Linux PC log
ping6 affe::585a:416d:6467:c752
PING affe::585a:416d:6467:c752(affe::585a:416d:6467:c752) 56 data bytes
64 bytes from affe::585a:416d:6467:c752: icmp_seq=1 ttl=63 time=34.6 ms
64 bytes from affe::585a:416d:6467:c752: icmp_seq=2 ttl=63 time=33.3 ms
64 bytes from affe::585a:416d:6467:c752: icmp_seq=3 ttl=63 time=33.3 ms
^C
--- affe::585a:416d:6467:c752 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 33.308/33.750/34.620/0.615 ms

cgundogan, 2016-03-01:
contratulations! you pinged from linux over the border router to the `gnrc_networking` node

girisugu2, 2016-03-01:
ok thanks alot.Its working for me now...So whats next?

girisugu2, 2016-03-01:
how i should communicate gnrc_border_router to gnrc_networking?

cgundogan, 2016-03-01:
What's your goal now? What kind of communication are you speaking of? Pinging works from the border example to the second node.

girisugu2, 2016-03-01:
yes.But i want to push messages to internet.So for that i need to send message from parent node to linux pc.So gnrc_border_router is intermediate one.i think it will send the messages to internet

cgundogan, 2016-03-01:
if you open a udp server on the `gnrc_networking` node with `udp server start 8888` and send a udp message from linux with `netcat -6u <ip address of  gnrc_networking node(starting with affe::)> 8888`, then it should work. The border router is just intermediary and forwards/translates the udp traffic.

girisugu2, 2016-03-01:
Thank you very much for your big help...It works fine for me.

With Regards,
R.Giritharan

cgundogan, 2016-03-01:
Glad I could help (:
Should we close this issue now?

[36 more replies]

closed 2016-03-02 by PeterKietzmann