RIOT-OS/RIOT #4858: Feature request: support IEEE 802.15.4 frame version 2

OlegHahm, 2016-02-20:
Since the IEEE 802.15.4-2015 standard is released a new frame format for 802.15.4, tagged as frame version 2, is defined. This is, e.g., required for the TSCH mode of 802.15.4 (or at least for 6TiSCH). Unfortunately, to the best of my knowledge, this frame version is not supported by any transceiver yet, so we would have to implement it in software. Since, AFAIK, the 2015 version of the standard is still not available for free, one might look into the [802.15.4e-2011](https://standards.ieee.org/findstds/standard/802.15.4e-2012.html) amendment.

gebart, 2017-09-10:
The standard has been made available for free at http://standards.ieee.org/about/get/802/802.15.html in case anyone is interested in taking on this task.

miri64, 2017-09-10:
With `gnrc_netif2` this can be handled GNRC-internally (as our frame building is at the moment as well), with a netopt to determine the frame version.

benpicco, 2020-06-30:
Sci-Hub has all the standards btw :wink:

maribu, 2022-09-16:
Ping?

maribu, 2023-05-17:
PingyMcPingface?

miri64, 2023-05-23:
@jia200x?

jia200x, 2023-05-23:
I think this is not relevant anymore, as openDSME and openWSN already handle the new versions.

maribu, 2023-05-23:
Then let's close this.

closed 2023-05-23 by maribu