RIOT-OS/RIOT #4872: improve real-time documentation, find coding conventions

kaspar030, 2016-02-21:
While it is certainly possible to create real-time applications with RIOT, depending on the definition and requirements, actual documentation and coding conventions are lacking.

We should come up with

a) conventions like "disabling IRQ's for less than N instructions is tolerable" (example)
b) document side effects caused by a)
c) document algorithmic implications on real-time requirements (e.g., "locking a mutex might traverse the list of currently locked threads")

...

LudwigKnuepfer, 2016-02-22:
> Maybe it's more purposeful to provide some tool for the task of identifying the possible side-effects anyways.

kaspar030, 2016-02-22:
> Maybe it's more purposeful to provide some tool for the task of identifying the possible side-effects anyways.

Interesting. You mean e.g., a static code analyzer beefed up to count critical section cycles?

LudwigKnuepfer, 2016-02-22:
Something like that, yes.

kYc0o, 2016-07-11:
any news on this? There's no milestone but a current issue is referencing this one, so maybe we need to put a milestone on this too.

LudwigKnuepfer, 2016-07-12:
Do as you please.

kaspar030, 2019-08-19:
Not much discussion here.

closed 2019-08-19 by kaspar030